start-jsk/rtmros_hironx #319: setTargetPoseRelative fails with any values

130s, 2015-02-06:
The last time I ran this method was in January when I didn't see this happen. Today I'm seeing this on 2 different Precise machine.

```
In [9]: robot.goInitial(tm=3)
[hrpsys.py] , JntAnglesOfGr=torso, INITPOSE[i]=[0], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=head, INITPOSE[i]=[0, 0], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=rarm, INITPOSE[i]=[-0.6, 0, -100, 15.2, 9.4, 3.2], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=larm, INITPOSE[i]=[0.6, 0, -100, -15.2, 9.4, -3.2], tm=3, wait=True
Out[9]: True

In [10]: robot.setTargetPoseRelative('larm', 'LARM_JOINT5', dz=0.01, tm=0.5)
[ 0.32557512  0.18236314  1.04762452] [-3.07343703 -1.56902262  3.07324704]
larm WAIST [0.32557512384154097, 0.18236314012331811, 1.0476245226774712] [-3.0734370292072337, -1.5690226177313638, 3.0732470442072231] 0.5
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[10]: False
```
IK error on simulator console:
```
IK Fail, iter = 50
  err : 5.67759e-08 ( -0.000228188 -1.61829e-05 6.66665e-05) < 1e-08
      : 1.50796e-09 ( 1.3689e-05 3.63397e-05 -1.64837e-08) < 1e-06
[setTargetPose] IK failed
```

Fails even with 0 displacement:
```
In [11]: robot.setTargetPoseRelative('larm', 'LARM_JOINT5', dz=0, tm=0.5)
[ 0.32557512  0.18236314  1.03762452] [-3.07343703 -1.56902262  3.07324704]
larm WAIST [0.32557512384154097, 0.18236314012331811, 1.0376245226774712] [-3.0734370292072337, -1.5690226177313638, 3.0732470442072231] 0.5
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[11]: False

[getCurrentPose] linkaname = LARM_JOINT5, frame_name = (null)
IK Fail, iter = 50
  err : 1.03758e-07 ( -0.000309656 -2.20198e-05 8.59463e-05) < 1e-08
      : 2.76389e-09 ( 1.84887e-05 4.92144e-05 -7.40257e-10) < 1e-06
[setTargetPose] IK failed
```

```
$ dpkg -p ros-hydro-hrpsys|grep Vers
Version: 315.2.7-0precise-20150101-1749-+0000
$ dpkg -p ros-hydro-hrpsys-ros-bridge|grep Vers
Version: 1.2.6-0precise-20150204-0145-+0000
$ dpkg -p ros-hydro-hironx-ros-bridge|grep Vers
Version: 1.0.27-0precise-20150204-0343-+0000
```

k-okada, 2015-02-06:
it works for me
```
k-okada@ubuntu:/opt/ros/hydro/share/hironx_ros_bridge/launch$ rtmlaunch  hironx_ros_bridge_simulation.launch 
[rtmlaunch] Start omniNames at port 15005 

Thu Feb  5 19:26:52 2015:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003137322e31362e3131372e32303000009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000007c34d45401000f21
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/k-okada/.ros/log/fe234f78-adaf-11e4-8193-000c2988acbc/roslaunch-ubuntu-3870.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:51103/

SUMMARY
========

PARAMETERS
 * /collision_state/comp_name
 * /controller_configuration
 * /diagnostic_aggregator/analyzers/computers/contains
 * /diagnostic_aggregator/analyzers/computers/path
 * /diagnostic_aggregator/analyzers/computers/type
 * /diagnostic_aggregator/analyzers/hrpsys/contains
 * /diagnostic_aggregator/analyzers/hrpsys/path
 * /diagnostic_aggregator/analyzers/hrpsys/type
 * /diagnostic_aggregator/analyzers/mode/contains
 * /diagnostic_aggregator/analyzers/mode/path
 * /diagnostic_aggregator/analyzers/mode/type
 * /diagnostic_aggregator/analyzers/motor/contains
 * /diagnostic_aggregator/analyzers/motor/path
 * /diagnostic_aggregator/analyzers/motor/type
 * /diagnostic_aggregator/base_path
 * /diagnostic_aggregator/pub_rate
 * /hironx/collada_model_filepath
 * /robot_description
 * /robot_pose_ekf/debug
 * /robot_pose_ekf/freq
 * /robot_pose_ekf/imu_used
 * /robot_pose_ekf/odom_used
 * /robot_pose_ekf/output_frame
 * /robot_pose_ekf/self_diagnose
 * /robot_pose_ekf/sensor_timeout
 * /robot_pose_ekf/vo_used
 * /rosdistro
 * /rosversion
 * /use_sim_time

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    base_footprint_rootlink (tf/static_transform_publisher)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys (hrpsys/hrpsys-simulator)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
[... 678 lines cut ...]
larm None [0.32556273687154708, 0.18236387337782681, 0.17462449717662004] [-3.0732189053889805, -1.5690225912054285, 3.0730289207320203] 0.5
Out[2]: True

In [3]: 
Do you really want to exit ([y]/n)? y
python: /usr/include/boost/thread/pthread/pthread_mutex_scoped_lock.hpp:26: boost::pthread::pthread_mutex_scoped_lock::pthread_mutex_scoped_lock(pthread_mutex_t*): Assertion `!pthread_mutex_lock(m)' failed.
Aborted (core dumped)
k-okada@ubuntu:/opt/ros/hydro/share/hironx_ros_bridge/scripts$ 

```
```
k-okada@ubuntu:/opt/ros/hydro/share/hironx_ros_bridge/scripts$ dpkg -p ros-hydro-hrpsys | grep Vers
Version: 315.2.7-0precise-20150101-1749-+0000
k-okada@ubuntu:/opt/ros/hydro/share/hironx_ros_bridge/scripts$ dpkg -p ros-hydro-hrpsys-ros-bridge | grep Vers
Version: 1.2.6-0precise-20150204-0145-+0000
k-okada@ubuntu:/opt/ros/hydro/share/hironx_ros_bridge/scripts$ dpkg -p ros-hydro-hironx-ros-bridge | grep Vers
Version: 1.0.27-0precise-20150204-0343-+0000
k-okada@ubuntu:/opt/ros/hydro/share/hironx_ros_bridge/scripts$ env | grep ROS
ROS_ROOT=/opt/ros/hydro/share/ros
ROS_PACKAGE_PATH=/opt/ros/hydro/share:/opt/ros/hydro/stacks
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
ROS_DISTRO=hydro
ROS_ETC_DIR=/opt/ros/hydro/etc/ros
```

130s, 2015-03-12:
Under the same version situation with #330, this doesn't happen with Hironx real robot.

closed 2015-03-12 by 130s