start-jsk/rtmros_hironx #320: hironx_tutorial depends on downstream

130s, 2015-02-06:
Continued from https://github.com/start-jsk/rtmros_hironx/commit/be3c9626a7365f87cf8e8d18405e13b30bd2b55a#commitcomment-9554559

@k-okada To summarize up, I think of either the following solution:
- Remove its dependency to `nextage*` packages
- Move to `rtmros_tutorial` (I'm for this option, but not sure if this makes sense)
- Separate as an individual package (as used to called as "unary stack" in rosbuild era)

I want to make a new release very soon. It's good if I can include the change to this issue.

k-okada, 2015-02-06:
On Fri, Feb 6, 2015 at 1:37 PM, Isaac I.Y. Saito <notifications@github.com>
wrote:

> Move to rtmros_tutorial (I'm for this option, but not sure if this makes
> sense)

+1 too

--
◉ Kei Okada

130s, 2015-02-20:
Fixed in https://github.com/start-jsk/rtmros_hironx/pull/321

closed 2015-02-20 by 130s