start-jsk/rtmros_hironx #330: "Wait for /hiroXXX:15005/sh.rtc:StateHolderService" lasts and eventually fail

130s, 2015-02-20:
```
roslaunch hironx_ros_bridge hironx_ros_bridge_real.launch nameserver:=hiro019
:
Wait for /hiroXXX:15005/sh.rtc:StateHolderService 1/30
Wait for /hiroXXX:15005/sh.rtc:StateHolderService 2/30
:
```
And fails.

However, a workaround is known as follows:
1. "ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host %HOSTNAME%"  --> Fails. So ends ipython session.
2. Run roslaunch  --> Should succeed.
3. Re-run ipython command above again.

The workaround above is not intuitive at all. Where should we implement the solution?

rtmros_hironx 1.0.27
rtmros_common 1.2.6

130s, 2015-03-12:
Now I don't see this with real robot (Hironx) with the following versions.
```
ros-hydro-hironx-calibration	1.0.28-0precise-20150206-2221-+0000
ros-hydro-hironx-moveit-config	1.0.28-0precise-20150311-1752-+0000
ros-hydro-hironx-ros-bridge	1.0.28-0precise-20150311-1725-+0000
ros-hydro-hrpsys	315.3.0-1precise-20150311-1614-+0000
ros-hydro-hrpsys-ros-bridge	1.2.8-0precise-20150311-1659-+0000
ros-hydro-hrpsys-tools	1.2.8-0precise-20150311-1654-+0000
ros-hydro-openhrp3	3.1.7-3precise-20150213-1403-+0000
ros-hydro-openrtm-aist	1.1.0-13precise-20150213-1339-+0000
ros-hydro-openrtm-aist-python	1.1.0-11precise-20150223-0443-+0000
ros-hydro-openrtm-tools	1.2.8-0precise-20150309-1040-+0000
ros-hydro-rtctree	3.0.3-2precise-20150304-0939-+0000
ros-hydro-rtmbuild	1.2.8-0precise-20150309-1054-+0000
ros-hydro-rtmros-hironx	1.0.28-0precise-20150311-1756-+0000
```

closed 2015-03-12 by 130s

k-okada, 2015-03-15:
```
roslaunch hironx_ros_bridge hironx_ros_bridge_real.launch 
nameserver:=hiro019
```

is this typo ? may be you need to use `rtmlaunch`

130s, 2015-03-15:
> is this typo ? may be you need to use `rtmlaunch`

No, this was when I worked with the real robot.