start-jsk/rtmros_hironx #331: TypeError: unbound method __init__() must be called with _objref_SDO instance as first argument (got _objref_DataFlowComponent instance instead)

130s, 2015-03-10:
```
$ rospack find hironx_ros_bridge
/opt/ros/indigo/share/hironx_ros_bridge
$ rosversion hironx_ros_bridge
1.0.28
$ dpkg -p ros-indigo-hironx-ros-bridge |grep Vers
Version: 1.0.28-0trusty-20150309-1142-+0000
$ ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host hiro
Python 2.7.6 (default, Mar 22 2014, 22:59:56) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with  hiro : 15005
[hrpsys.py]  waiting ModelLoader
[hrpsys.py]  start hrpsys
[hrpsys.py]  finding RTCManager and RobotHardware
[hrpsys.py]  wait for RTCmanager :  None
omniORB: Caught an unexpected Python exception during up-call.
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM_idl.py", line 48, in __init__
    _0_RTC._objref_RTObject.__init__(self)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1341, in __init__
    _0_SDOPackage._objref_SDO.__init__(self)
TypeError: unbound method __init__() must be called with _objref_SDO instance as first argument (got _objref_DataFlowComponent instance instead)

If you suspect this is an IPython bug, please report it at:
    https://github.com/ipython/ipython/issues
or send an email to the mailing list at ipython-dev@scipy.org

You can print a more detailed traceback right now with "%tb", or use "%debug"
to interactively debug it.

Extra-detailed tracebacks for bug-reporting purposes can be enabled via:
    %config Application.verbose_crash=True

omniORB: Caught an unexpected Python exception during up-call.
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM_idl.py", line 48, in __init__
    _0_RTC._objref_RTObject.__init__(self)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1341, in __init__
    _0_SDOPackage._objref_SDO.__init__(self)
TypeError: unbound method __init__() must be called with _objref_SDO instance as first argument (got _objref_DataFlowComponent instance instead)

If you suspect this is an IPython bug, please report it at:
    https://github.com/ipython/ipython/issues
or send an email to the mailing list at ipython-dev@scipy.org

You can print a more detailed traceback right now with "%tb", or use "%debug"
to interactively debug it.

Extra-detailed tracebacks for bug-reporting purposes can be enabled via:
    %config Application.verbose_crash=True

[hrpsys.py]  wait for RobotHardware0  :  None (timeout  0  < 10)
omniORB: Caught an unexpected Python exception during up-call.
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/OpenRTM_idl.py", line 48, in __init__
    _0_RTC._objref_RTObject.__init__(self)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1341, in __init__
    _0_SDOPackage._objref_SDO.__init__(self)
TypeError: unbound method __init__() must be called with _objref_SDO instance as first argument (got _objref_DataFlowComponent instance instead)

If you suspect this is an IPython bug, please report it at:
    https://github.com/ipython/ipython/issues
or send an email to the mailing list at ipython-dev@scipy.org

[... 116 lines cut ...]
/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py in waitForRobotHardware(self, robotname)
    757             print self.configurator_name, "Could not find ", robotname
    758             if self.ms:
--> 759                 print self.configurator_name, "Candidates are .... ", [x.name()  for x in self.ms.get_components()]
    760             print self.configurator_name, "Exitting.... ", robotname
    761             exit(1)

/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/rtm.py in get_components(self)
    226     def get_components(self):
    227         cs = []
--> 228         crefs = self.ref.get_components()
    229         for cref in crefs:
    230             c = RTcomponent(cref)

/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/Manager_idl.pyc in get_components(self, *args)
    163 
    164     def get_components(self, *args):
--> 165         return _omnipy.invoke(self, "get_components", _0_RTM.Manager._d_get_components, args)
    166 
    167     def get_component_profiles(self, *args):

UNKNOWN: CORBA.UNKNOWN(omniORB.UNKNOWN_PythonException, CORBA.COMPLETED_MAYBE)

In [1]: 
```

130s, 2015-03-10:
This error doesn't occur when I purge openrtm-aist (which unistalls many downstream packages too), then re-installed from non shadow-fixed repo rtmros-hironx (which installs many upstream pkgs).

130s, 2015-03-10:
Although not sure exactly where is the root cause, [I asked ROS team to hold the public sync](http://lists.ros.org/pipermail/ros-release/2015-March/004593.html), only for Hydro (it is happening with Indigo too but we haven't even announced the maintenanceship for Indigo yet).

k-okada, 2015-03-10:
it is always good to check test code in  upstream package to find the cause, it happens on hrpsys test code, so it should be hrpsys problem, nor rtmros_hironx problem
move to https://github.com/fkanehiro/hrpsys-base/pull/506

closed 2015-03-10 by k-okada

130s, 2015-03-10:
I'd be sorry if I missed something to do in downstream packages. Please let me know if there's anything needs done in this repo.

Because I haven't seen this issue on simulation until yesterday using shadow-fixed, it should be rooted in the very new change(s).

130s, 2015-03-11:
With the following version available in shadow-fixed, this issue is gone on simulation.
```
$ dpkg -p ros-hydro-hrpsys|grep Ver
Version: 315.3.0-1precise-20150311-1614-+0000
```

130s, 2015-03-11:
Same for Indigo. I think we can unblock the public sync hold. Thank you x 10 @k-okada for the quick fix.
```
$ dpkg -p ros-indigo-hrpsys |grep Vers
Version: 315.3.0-1trusty-20150311-1141-+0000
```