start-jsk/rtmros_hironx #333: setTargetPoseRelative always fails

130s, 2015-03-17:
Starting the last few weeks, even the simplest relative movements like these (dx=0.01 or dz=0.01, 0.001 etc.) fail.
```
In [2]: robot.setTargetPoseRelative('rarm', 'RARM_JOINT5', dx=0.01, tm=0.5, wait=False)
[ 0.33484857 -0.18201862  1.03680264] [ 0.17900024 -1.57002806 -0.17776429]
rarm WAIST [0.33484856729528889, -0.18201862397166299, 1.0368026417623699] [0.17900023850191049, -1.5700280609383033, -0.17776428666024424] 0.5
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[2]: False

In [3]: robot.setTargetPoseRelative('rarm', 'RARM_JOINT5', dz=0.01, tm=0.5, wait=False)
[ 0.32484857 -0.18201862  1.04680264] [ 0.17900024 -1.57002806 -0.17776429]
rarm WAIST [0.32484856729528888, -0.18201862397166299, 1.0468026417623699] [0.17900023850191049, -1.5700280609383033, -0.17776428666024424] 0.5
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
(like the one in https://github.com/start-jsk/rtmros_hironx/issues/103) is not implemented. Patch is welcomed.
Out[3]: False
```
```
$ dpkg -p ros-indigo-openhrp3 ros-indigo-hrpsys ros-indigo-rtmros-common ros-indigo-rtmros-hironx ros-indigo-rtmros-nextage|grep Vers
Version: 3.1.7-15trusty-20150213-1223-+0000
Version: 315.3.0-1trusty-20150311-1141-+0000
Version: 1.2.8-0trusty-20150311-1229-+0000
Version: 1.0.28-0trusty-20150311-1244-+0000
Version: 0.6.0-0trusty-20150311-1247-+0000
```

k-okada, 2015-03-17:
please create test code as https://github.com/start-jsk/rtmros_hironx/pull/332

k-okada, 2015-03-17:
create test at #334,

130s, 2015-03-28:
On simulator I confirmed this issue can be worked around after running `goInitial` once.

Permanent fix https://github.com/fkanehiro/hrpsys-base/pull/519.

130s, 2015-03-28:
Using the following, this issue still happens. Running `goInitial` doesn't solve.

```
$ dpkg -p ros-indigo-hrpsys|grep Ver
Version: 315.3.0-1trusty-20150323-1010-+0000
```
[Latest commit of hrpsys-base](https://github.com/fkanehiro/hrpsys-base/commit/15c5b8c1027954b64ffbc7eb0c40e01ff1e48221)

130s, 2015-03-28:
> On simulator I confirmed this issue can be worked around after running goInitial once.

When I said the above, I forgot to get rid of the [temporary commit](https://github.com/fkanehiro/hrpsys-base/pull/518) (that's now closed).

130s, 2015-03-29:
I found with hironx this works. It fails only with nextage. Moved to https://github.com/tork-a/rtmros_nextage/issues/160

closed 2015-03-29 by 130s