start-jsk/rtmros_hironx #335: yaw movement with setTargetPoseRelative actually moves roll

130s, 2015-03-17:
See [the youtube video](https://www.youtube.com/watch?v=dgHBdr4ha6Y&feature=youtu.be); Both `roll` movement that comes as the 1st batch, and `yaw`, 3rd, rotate about red axis (x) while we expect `yaw` rotates about blue axis.

Result of getCurrentRPY('RARM_JOINT5') indicates that `yaw` is rotated as expected in the 3rd batch of the movements.

```
$ ipython -i ~/Desktop/nextage_RPY_position_test.py 
:
[hrpsys.py]    setupLogger :  emergencySignal  arleady exists in DataLogger
HiroNX(Robot)0.emergencySignal and log.emergencySignal are already connected
[hrpsys.py]  setup joint groups
[hrpsys.py]  initialized successfully
[hrpsys.py] , JntAnglesOfGr=torso, INITPOSE[i]=[0], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=head, INITPOSE[i]=[0, 0], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=rarm, INITPOSE[i]=[-0.6, 0, -100, 15.2, 9.4, 3.2], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=larm, INITPOSE[i]=[0.6, 0, -100, -15.2, 9.4, -3.2], tm=3, wait=True
goInitial (0.1790002385019105, -1.5700280609383033, -0.17776428666024424) (-0.1790002385019105, -1.5700280609383033, 0.17776428666024424)
[ 0.32484857 -0.18201862  1.03680264] [ 1.74979657 -1.57002806 -0.17776429]
rarm None [0.32484856729528888, -0.18201862397166299, 1.0368026417623699] [1.749796565296807, -1.5700280609383033, -0.17776428666024424] 0.5
[ 0.32484857  0.18201862  1.03680264] [ 1.39179609 -1.57002806  0.17776429]
larm None [0.32484856729528888, 0.18201862397166299, 1.0368026417623699] [1.3917960882929861, -1.5700280609383033, 0.17776428666024424] 0.5
Roll +move  (0.1790002385019105, -1.5700280609383033, -0.17776428666024424) (-0.1790002385019105, -1.5700280609383033, 0.17776428666024424)
[ 0.32484839 -0.1820044   1.03680268] [ 0.15324298 -1.57002812 -0.17760291]
rarm None [0.32484838559554136, -0.18200440398211598, 1.0368026803470394] [0.15324298212371934, -1.5700281203401243, -0.17760290598143333] 0.5
[ 0.32484863  0.18203201  1.03680262] [-0.21436701 -1.57002801  0.17790798]
larm None [0.32484863037008305, 0.18203200857881313, 1.0368026224539324] [-0.21436700764251593, -1.5700280103287043, 0.17790797525715421] 0.5
Roll -move (1.744656077208585, -1.5700281208977755, -0.17760154578253698) (1.3660562556193259, -1.5700280101240005, 0.1779084821051138)
[ 0.3248485  -0.18201442  1.03680263] [ 1.74974638 -1.57002808 -0.17771547]
rarm None [0.32484849621018302, -0.18201442349228358, 1.0368026254281859] [1.7497463834078084, -1.5700280822997297, -0.17771546909653543] 0.5
[ 0.32484838  0.18202196  1.03680271] [ 1.3909327  -1.57002805  0.17778358]
larm None [0.32484838182331555, 0.18202196252587466, 1.0368027058128837] [1.3909326953942442, -1.5700280546272203, 0.17778358010850423] 0.5
Roll +move  (0.15361466516427447, -1.5700280823072363, -0.17771550127346952) (-0.2098782738823774, -1.570028054790546, 0.17778324845150964)
[ 0.32484831 -0.18200027  1.03680266] [ 0.1438193  -1.57002814 -0.17755467]
rarm None [0.3248483131389025, -0.18200027174288302, 1.0368026648250896] [0.14381929862865706, -1.5700281414490111, -0.1775546713947975] 0.5
[ 0.32484844  0.18203526  1.03680269] [-0.20534076 -1.570028    0.17792629]
larm None [0.32484844344799041, 0.18203526076210838, 1.0368026872556013] [-0.20534075613557379, -1.570028004328855, 0.17792629438896618] 0.5
Roll -move (1.7245984660662992, -1.570028105203389, -0.177660156817488) (1.3654555706593228, -1.570028004328855, 0.17792629438896618)
[ 0.32484842 -0.18201019  1.03680261] [ 1.74910064 -1.5700281  -0.17766615]
rarm None [0.32484842287336058, -0.18201019020085168, 1.0368026103862076] [1.7491006416851793, -1.5700281037703667, -0.17766615262010349] 0.5
[ 0.32484819  0.18202512  1.03680277] [ 1.39117159 -1.57002805  0.17780066]
larm None [0.32484819260901748, 0.18202511642228059, 1.0368027713795751] [1.3911715872613182, -1.570028049083094, 0.17780066464986891] 0.5
Roll +move  (0.14418737574345405, -1.5700281037840633, -0.1776662111497786) (-0.20052820181824185, -1.5700280491913066, 0.17780043424131944)
[ 0.32484824 -0.18199612  1.03680265] [ 0.15281004 -1.57002816 -0.17750644]
rarm None [0.32484824262258183, -0.18199611757921039, 1.0368026486901556] [0.15281004220255756, -1.570028162532829, -0.1775064419389521] 0.5
[ 0.32484825  0.18203855  1.03680275] [-0.20536445 -1.570028    0.1779448 ]
larm None [0.32484825469905326, 0.18203855159740079, 1.036802752661967] [-0.20536444984092772, -1.5700279982759591, 0.17794480488449974] 0.5
Roll -move (1.7486814350099888, -1.5700281632060062, -0.17750479892282378) (1.3907658049943628, -1.570027997720863, 0.177946187132938)
[ 0.32484835 -0.18200614  1.03680259] [ 1.7493093  -1.57002812 -0.17761901]
rarm None [0.32484835349148611, -0.18200613574410668, 1.0368025937198257] [1.7493093016035381, -1.570028124508033, -0.17761901073279637] 0.5
[ 0.324848    0.18202842  1.03680284] [ 1.40052605 -1.57002804  0.17781926]
larm None [0.32484800372723144, 0.1820284178900532, 1.0368028369100606] [1.4005260532154342, -1.5700280429911573, 0.17781926340213383] 0.5
Roll +move  (0.1531816882957567, -1.5700281245153878, -0.17761904224871813) (-0.20055112205743592, -1.570028043147831, 0.17781893043266497)
[ 0.32484817 -0.18199199  1.03680263] [ 0.1433823  -1.57002818 -0.1774582 ]
rarm None [0.32484816998453381, -0.18199198633981861, 1.0368026332326989] [0.14338229718932127, -1.5700281836348902, -0.17745820261119491] 0.5
[ 0.32484806  0.18204178  1.03680282] [-0.19575723 -1.57002799  0.17796252]
larm None [0.32484806433244667, 0.18204177573518329, 1.0368028189798939] [-0.19575722686942232, -1.5700279924647136, 0.17796251977462324] 0.5
Roll -move (1.748885705847304, -1.5700281845670165, -0.17745592746427152) (1.3960171852178769, -1.5700279920018114, 0.17796367456448128)
[ 0.32484828 -0.18200191  1.03680258] [ 1.73963271 -1.57002815 -0.17756972]
rarm None [0.32484827996384458, -0.18200190647282274, 1.0368025787498745] [1.7396327067945327, -1.5700281459691199, -0.17756972174664676] 0.5
[ 0.32484781  0.18203164  1.0368029 ] [ 1.40075514 -1.57002804  0.17783701]
larm None [0.32484781350733644, 0.18203164324825119, 1.0368029031909454] [1.4007551442652795, -1.5700280371805302, 0.17783700551888248] 0.5
Roll +move  (0.1437503966422288, -1.570028145979322, -0.177569764698699) (-0.1953723585159133, -1.5700280373120055, 0.1778367265153489)
[ 0.3248481  -0.18198783  1.03680262] [ 0.13371171 -1.5700282  -0.17740952]
rarm None [0.32484809511886348, -0.18198782733150745, 1.0368026190667539] [0.13371170595743509, -1.5700282048348182, -0.17740952058854015] 0.5
[ 0.32484787  0.18204526  1.03680289] [-0.17487557 -1.57002799  0.1779828 ]
larm None [0.32484787277885863, 0.18204525919403394, 1.0368028864405689] [-0.17487557442056945, -1.5700279856950057, 0.17798280224114096] 0.5
Roll -move (1.7392098331081038, -1.570028205745979, -0.17740729613445955) (1.3959207523743271, -1.5700279856950057, 0.17798280224114096)

[hrpsys.py] , JntAnglesOfGr=torso, INITPOSE[i]=[0], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=head, INITPOSE[i]=[0, 0], tm=3, wait=True
[hrpsys.py] , JntAnglesOfGr=rarm, INITPOSE[i]=[-0.6, 0, -100, 15.2, 9.4, 3.2], tm=3, wait=True
[... 132 lines cut ...]
robot.goInitial(3)
print "goInitial", robot.getCurrentRPY('RARM_JOINT5'), robot.getCurrentRPY('LARM_JOINT5')

#---------------------pitch motion-------------------------------------------------------------------
for j in range(0, 5):
	robot.setTargetPoseRelative('rarm', 'RARM_JOINT5', dp=math.pi/4, tm=0.5, wait=False)
	robot.setTargetPoseRelative('larm', 'LARM_JOINT5', dp=math.pi/4, tm=0.5)
	print "Pitch +move", robot.getCurrentRPY('RARM_JOINT5'), robot.getCurrentRPY('LARM_JOINT5')
	robot.setTargetPoseRelative('rarm', 'RARM_JOINT5', dp=-math.pi/4, tm=0.5, wait=False)
	robot.setTargetPoseRelative('larm', 'LARM_JOINT5', dp=-math.pi/4, tm=0.5)
	print "Pitch -move", robot.getCurrentRPY('RARM_JOINT5'), robot.getCurrentRPY('LARM_JOINT5')
robot.goInitial(3)
print "goInitial", robot.getCurrentRPY('RARM_JOINT5'), robot.getCurrentRPY('LARM_JOINT5')

#---------------------yaw motion---------------------------------------------------------------------
for j in range(0, 5):
	robot.setTargetPoseRelative('rarm', 'RARM_JOINT5', dw=math.pi/2, tm=0.5, wait=False)
	robot.setTargetPoseRelative('larm', 'LARM_JOINT5', dw=math.pi/2, tm=0.5)
	print "Yaw +move", robot.getCurrentRPY('RARM_JOINT5'), robot.getCurrentRPY('LARM_JOINT5')
	robot.setTargetPoseRelative('rarm', 'RARM_JOINT5', dw=-math.pi/2, tm=0.5, wait=False)
	robot.setTargetPoseRelative('larm', 'LARM_JOINT5', dw=-math.pi/2, tm=0.5)
	print "Yaw -move", robot.getCurrentRPY('RARM_JOINT5'), robot.getCurrentRPY('LARM_JOINT5')
robot.goInitial(3)
print "goInitial", robot.getCurrentRPY('RARM_JOINT5'), robot.getCurrentRPY('LARM_JOINT5')
```

130s, 2015-03-17:
#BTW, the video also shows that with `pitch` movement the y axis (green) is moving while it's not supposed to. This maybe related to https://github.com/start-jsk/rtmros_hironx/issues/287

130s, 2015-03-17:
Used the following. I had to apply [this patch](https://github.com/fkanehiro/hrpsys-base/pull/518) for hrpsys.
```
$ dpkg -p ros-indigo-openhrp3 ros-indigo-hrpsys ros-indigo-rtmros-common ros-indigo-rtmros-hironx ros-indigo-rtmros-nextage|grep Vers
Version: 3.1.7-15trusty-20150213-1223-+0000
Version: 315.3.0-1trusty-20150311-1141-+0000
Version: 1.2.8-0trusty-20150311-1229-+0000
Version: 1.0.28-0trusty-20150311-1244-+0000
Version: 0.6.0-0trusty-20150311-1247-+0000
```

k-okada, 2015-03-17:
it seems relative means relative respect to RARM_LINK5 local coordinates and the convention is
```
red = x = roll
green = y = pitch
blue = z = yaw
```
so this seems ok.
if you wan to move RARM_LINK5, relative to world coordinates, may be you can add a few line to hrpsys_config.py,

also please look at https://github.com/start-jsk/rtmros_hironx/blob/hydro-devel/hironx_ros_bridge/test/test_hironx_cartesian.py#L88 to learn how to write test code.

130s, 2015-03-23:
> it seems relative means relative respect to RARM_LINK5 local coordinates and the convention is
> ```
> red = x = roll
> green = y = pitch
> blue = z = yaw
> ```
> so this seems ok.

Thank you for the explanation. 

What about `blue = z = yaw`? In the video yaw movement didn't happen while it was supposed to. Instead roll moved again.

emijah, 2015-03-23:
This is because of how the RPY gets calculated. When the elbow is rotated 90 degrees the rotation itself can become ambiguous.

This is the reason I was trying to push the setTargetPoseMatrix interface. We've implemented this using python and are happy with this for now.

k-okada, 2015-03-23:
I see, 

something like
```            posRef = numpy.array([tds.data[3], tds.data[7], 
tds.data[11]])
            matRef = numpy.array([tds.data[0:3], tds.data[4:7], 
tds.data[8:11]])
            posRef += [dx, dy, dz]
            matRef = matRef.dot(numpy.array(euler_matrix(dr,dp,dw)[:3,:3]))
            rpyRef = euler_from_matrix(matRef)
            print posRef, rpyRef
            ret = self.setTargetPose(gname, list(posRef), list(rpyRef), tm)
```
may solve the problem

130s, 2015-04-01:
Addressed in the upstream https://github.com/fkanehiro/hrpsys-base/issues/543.

closed 2015-04-01 by 130s