start-jsk/rtmros_hironx #337: (ImpedanceController) TypeError: __init__() got an unexpected keyword argument 'ref_moment'

130s, 2015-03-17:
```
In [3]: robot.startImpedance(arm='rarm')
:
----> 1 robot.startImpedance(arm='rarm')

/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.pyc in startImpedance(self, arm, M_p, D_p, K_p, M_r, D_r, K_r, ref_force, force_gain, ref_moment, moment_gain, sr_gain, avoid_gain, reference_gain, manipulability_limit)
    667                 avoid_gain = avoid_gain,
    668                 reference_gain = reference_gain,
--> 669                 manipulability_limit = manipulability_limit))
    670 
    671     def stopImpedance(self, arm):

TypeError: __init__() got an unexpected keyword argument 'ref_moment'
```

Happening at [OpenHRP.ImpedanceControllerService](https://github.com/start-jsk/rtmros_hironx/blob/6467e8e8f2867ec185a34b4a2697260fb860ddb2/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L652)
 ```
 OpenHRP.ImpedanceControllerService.impedanceParam(
:
ref_moment = ref_moment,
:
```

Is this related to the [removal of ref_moment in hrpsys-base](https://github.com/fkanehiro/hrpsys-base/commit/fb89ad0ac678ca7ae0c913883fc17fc9a5f64777#diff-7100325148afb9fedc80b245672a9f49L38) recently happened?

```
$ dpkg -p ros-hydro-openhrp3 ros-hydro-hrpsys ros-hydro-rtmros-common ros-hydro-rtmros-hironx |grep Ver
Version: 3.1.7-3precise-20150213-1403-+0000
Version: 315.3.0-1precise-20150311-1614-+0000
Version: 1.2.8-0precise-20150311-1711-+0000
Version: 1.0.28-0precise-20150311-1756-+0000
```

k-okada, 2015-03-17:
you must check hrpsys version of QNX, may be you need update or use old
hrpsys_ros_bridge, that matches version of hrpsys

k-okada, 2015-03-17:
Oh, this happens on simulator, so this is hrpsys-base problem, not hiornx

k-okada, 2015-03-17:
oh, this is my misunderstanding, this is not hrpsys-base problem, the problem and workaround is proposed here -> https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/SampleRobot.conf.in#L5

130s, 2015-04-03:
> this is not hrpsys-base problem, the problem and workaround is proposed here -> https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/SampleRobot.conf.in#L5

Tried to understand this, but couldn't find anywhere [SampleRobot.conf](https://github.com/fkanehiro/hrpsys-base/blob/17738f8783156421db236b463563e8a9e2e12ecc/sample/SampleRobot/SampleRobot.conf.in#L5) (or its derivative) is in use, so I still don't know how this `ref_moment` issue can be solved.

130s, 2015-04-08:
Looks like fixed in https://github.com/start-jsk/rtmros_hironx/pull/351 thanks to @k-okada

closed 2015-04-08 by 130s