start-jsk/rtmros_hironx #345: How to get feedback values from the impedance RTC?

130s, 2015-03-27:
Sorry if I'm taking something wrong, but how to get feedback values from the impedance RTC in python/`HIRONX`? 
I see how to start the RTC (by using [startImpedance](https://github.com/start-jsk/rtmros_hironx/blob/37eae39294ada6146282f0f8286a78e88280e5dc/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L623)). I think necessary change is made on QNX. Thank you.
@k-okada @hyaguchijsk

k-okada, 2015-03-27:
what do you mean by "feedback value"? force sensor value? then you can
check with output of impedance rtc, such as lhsensor/rhsensor

130s, 2015-03-27:
I see. Then, apologies for another elementary question. I understand that `ImpedanceControllerService` can be accessed via `self.ic_svc` in HIRONX class, but [in the API doc](http://fkanehiro.github.io/hrpsys-base/d9/d8b/interfaceOpenHRP_1_1ImpedanceControllerService.html) I don't find any methods that return force sensor feedback.

k-okada, 2015-03-27:
sorry to ask again, what do you mean by "force sensor feedback" ? "value of force sensor output"? or "parameters of force sensor feedback controller"?

130s, 2015-03-27:
Yes, "value of force sensor output".

k-okada, 2015-03-27:
Ok, so, that is publish via openrtm port so we do not have api to get that value, use rtprint or rostopic echo

130s, 2015-03-27:
Thank you. 
One last question: what can the topic name be? I think [the topic in question might be published in HrpsysSeqStateROSBridge.cpp](https://github.com/start-jsk/rtmros_common/blob/826577a17e6854ac8dc2d4a1d149a8e3240a9eb5/hrpsys_ros_bridge/src/HrpsysSeqStateROSBridge.cpp#L132) but the topic name is variable. Prefix of the topic name should be enough (I don't have an access to the real robot where Impedance Controller runs).

k-okada, 2015-03-27:
thats comes from robot model definition,
https://github.com/start-jsk/rtmros_common/blob/826577a17e6854ac8dc2d4a1d149a8e3240a9eb5/hrpsys_ros_bridge/src/HrpsysSeqStateROSBridgeImpl.cpp#L125

so it should be lhsensor
https://github.com/start-jsk/rtmros_hironx/commit/a6a09ca6351d7e2295a59561eb81bfa0e4b21c41#diff-9263df30885d75490d8648c8453b3b18

130s, 2015-03-27:
Thank you! [Added to the tutorial](http://wiki.ros.org/action/info/rtmros_nextage/Tutorials/Adding%20feedback%20control?action=diff&rev2=9&rev1=8).

closed 2015-03-27 by 130s