start-jsk/rtmros_hironx #348: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

130s, 2015-03-29:
Must be related to https://github.com/rdiankov/collada-dom/pull/12
I'm afraid `tf`-based applications doesn't function at all.

```
$ rtmlaunch hironx_ros_bridge hironx_ros_bridge_simulation.launch
:
[ INFO] [1427586289.914961258]: [HrpsysSeqStateROSBridge] @Initilize name : HrpsysSeqStateROSBridge0
cache found for /opt/ros/indigo/share/nextage_description/models/main.wrl
Loading body customizer "/opt/ros/indigo/share/OpenHRP-3.1/customizer/libSampleCustomizer.so" for springJoint
cache found for /opt/ros/indigo/share/nextage_description/models/main.wrl
duplicated sensor Id is specified(id = 0, name = CAMERA_HEAD_R)
duplicated sensor Id is specified(id = 1, name = CAMERA_HEAD_L)
[ INFO] [1427586289.923313354]: [HrpsysSeqStateROSBridge] Loaded HiroNX
[ INFO] [1427586289.923722204]: [HrpsysSeqStateROSBridge] @Initilize name : HrpsysSeqStateROSBridge0 done
process[hrpsys_ros_diagnostics-8]: started with pid [1163]
configuration ORB with  localhost : 15005
[hrpsys.py]  waiting ModelLoader
[hrpsys.py]  start hrpsys
[hrpsys.py]  finding RTCManager and RobotHardware
process[diagnostic_aggregator-9]: started with pid [1172]
[ERROR] [1427586290.288093690]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

[ERROR] [1427586290.304585212]: COLLADA error: Failed to load XML document from memory

state_publisher: /usr/include/boost/smart_ptr/shared_ptr.hpp:653: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const urdf::Link; typename boost::detail::sp_member_access<T>::type = const urdf::Link*]: Assertion `px != 0' failed.
[hrpsys_state_publisher-7] process has died [pid 1095, exit code -6, cmd /opt/ros/indigo/lib/robot_state_publisher/state_publisher __name:=hrpsys_state_publisher __log:=/home/n130s/.ros/log/6b13a468-d5a4-11e4-9fc6-f816542d218e/hrpsys_state_publisher-7.log].
log file: /home/n130s/.ros/log/6b13a468-d5a4-11e4-9fc6-f816542d218e/hrpsys_state_publisher-7*.log
process[hrpsys_profile-10]: started with pid [1274]
/opt/ros/indigo/share/hrpsys_ros_bridge/scripts/diagnostics.py:75: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
```

`robot_state_publisher` is not alive.
```
$ rosnode list
/DataLoggerServiceROSBridge
/ForwardKinematicsServiceROSBridge
/HrpsysJointTrajectoryBridge
/HrpsysSeqStateROSBridge
/SequencePlayerServiceROSBridge
/StateHolderServiceROSBridge
/base_footprint_rootlink
/collision_state
/diagnostic_aggregator
/hrpsys_profile
/hrpsys_ros_diagnostics
/hrpsys_state_publisher
/robot_pose_ekf
/rosout
```
[image]

This happens with `rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch` as well. tf tree looks slightly different as follows.

[image]

Not sure when this issue started happening, but at least in [this ticket from Novemeber 2014](https://github.com/start-jsk/rtmros_hironx/issues/287), tf was working on simulator.

```
n130s@tork-kudu1:~$ dpkg -p collada-dom2.4-dp-base | grep Ver
Version: 2.4.3.1-ubuntu1~trusty1
n130s@tork-kudu1:~$ dpkg -p collada-dom2.4-dp-dev | grep Ver
Version: 2.4.3.1-ubuntu1~trusty1
n130s@tork-kudu1:~$ dpkg -p collada-dom-dev | grep Ver
Version: 2.4.3.1-ubuntu1~trusty1
n130s@tork-kudu1:~$ dpkg -p ros-indigo-hrpsys ros-indigo-openhrp3 ros-indigo-hironx-ros-bridge ros-indigo-nextage-ros-bridge | grep Ver
Version: 315.3.0-1trusty-20150323-1010-+0000
Version: 3.1.7-15trusty-20150213-1223-+0000
Version: 1.0.28-0trusty-20150323-1855-+0000
Version: 0.6.0-0trusty-20150324-0008-+0000
```

k-okada, 2015-03-29:
> Must be related to rdiankov/collada-dom#12

indigo is not ready yet.

130s, 2015-03-29:
I see, then appointed to [Indigo release milestone](https://github.com/start-jsk/rtmros_hironx/milestones/Indigo%20release).

130s, 2015-05-11:
This no longer happens on the same Trusty-Indigo machine.

```
$ dpkg -p collada-dom2.4-dp-base | grep Ver
Version: 2.4.3.1-ubuntu1~trusty1
$ dpkg -p collada-dom2.4-dp-dev | grep Ver
Version: 2.4.3.1-ubuntu1~trusty1
$ dpkg -p collada-dom-dev | grep Ver
Version: 2.4.3.1-ubuntu1~trusty1
$ dpkg -p ros-indigo-hrpsys ros-indigo-openhrp3 ros-indigo-hironx-ros-bridge ros-indigo-nextage-ros-bridge | grep Ver
Version: 315.4.0-0trusty-20150507-0829-+0000
Version: 3.1.8-0trusty-20150424-0521-+0000
Version: 1.0.30-0trusty-20150507-1540-+0000
Version: 0.6.0-0trusty-20150507-1757-+0000
```
Looks like the only different packages are `ros-indigo-*`, not `collada-*`.

closed 2015-05-11 by 130s