start-jsk/rtmros_hironx #357: Compensate hands weight from python script?

130s, 2015-04-15:
A user is experiencing that hands don't stay still when the custom heavier hands are attached. Is there a user command to compensate weight?

130s, 2015-04-16:
Title changed:
Old: Compensate hands weight from python script?
New: Compensate hands weight?

I think this doesn't need to be done via python (if that's possible it'd be very convenient though).

Any idea folks?

130s, 2015-04-16:
In ROS there's a [gravity_compensation](http://wiki.ros.org/gravity_compensation) package. In order to utilize it, there are [some prerequisites](http://wiki.ros.org/force_torque_tools/Tutorials/Force-torque%20sensor%20calibration#Overview_and_prerequisites) that Hironx + JR3 device might not satisfy though.

k-okada, 2015-04-16:
since the gripper of this robot usually facing down to the table, so
it is better to start from simple one (and always it is better to
start from simple case rather than general case)
jr3 drive has feature to cancel the offset
https://github.com/start-jsk/rtmros_hironx/blob/hydro-devel/hironx_ros_bridge/robot/nitta/jr3_driver.cpp#L508
and I though that rtcd cancel offset when it started, but it was
wrong, so we need add code to send that command to robot
https://github.com/start-jsk/rtmros_hironx/pull/253

```

+   /* Clear the memory for the msg and the reply */
+   memset (&msg, 0, sizeof (msg));
+   memset (&msg_reply, 0, sizeof (msg_reply));
+
+   /* Setup the message data to send to the server */
+   num = 4;
+   msg.msg_no = _IO_MAX + num;
+   snprintf (msg.msg_data, 254, "client %d requesting reply.", getpid ());
+
+   printf ("client: msg_no: _IO_MAX + %d\n", num);
+   fflush (stdout);
+
+   ret = MsgSend (force_sensor_fd, &msg, sizeof (msg), msg_reply, 255);
+   if (ret == -1)
+     {
+ fprintf (stderr, "Unable to MsgSend() to server: %s\n",
+ strerror (errno));
+ return EXIT_FAILURE;
+     }
+   printf ("client: msg_reply:\n%s\n", msg_reply);

```

may be I removed this code since when rtcd start, usually the hand
position is not good for calibration,
may be send '2' at
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/RobotHardware/robot.cpp#L293
when servo is on, is better

next step is to use following rtc, which take care of this feature. it
does not require source compile, just to change startup script
https://github.com/fkanehiro/hrpsys-base/tree/315.1.10/rtc/AbsoluteForceSensor

you have to compensate the weight of hand within the controller, so
any ros package is not applicable.
--
◉ Kei Okada

130s, 2015-04-20:
I say in advance that I'm sorry to ask many questions, but this is very much complicated to me.

> may be I removed this code since when rtcd start, usually the hand position is not good for calibration, 

The code cited [is in sample.c](https://github.com/start-jsk/rtmros_hironx/blob/b499aad046f0e4818a798a033cbd9b576fde9605/hironx_ros_bridge/robot/nitta/sample.c#L70).

> may be send '2' at https://github.com/fkanehiro/hrpsys-base/blob/47cf3605ad409585aee23a9e7aa1e290471247fd/rtc/RobotHardware/robot.cpp#L293 when servo is on, is better.

Could you tell me how to do so? 

And are you referring to the right line?

```
292    if (strcmp(jname, "all") == 0 || strcmp(jname, "ALL") == 0){
293        bool ret = true;
294        for (int i=0; i<numJoints(); i++){
```

> next step is to use following rtc, which takes care of this feature. it does not require source compile, just to change startup script https://github.com/fkanehiro/hrpsys-base/tree/315.1.10/rtc/AbsoluteForceSensor

I have no idea how to do so.
Isn't [this line](https://github.com/start-jsk/rtmros_hironx/commit/f551037aa697c1cf8b0faf1291e4c3565eba0059#diff-f9592097d70ddcac8ad91644f3717047R199) for example doing what you meant??

k-okada, 2015-04-21:
Sorry for unhelpful comment

My thought was put

+   /* Clear the memory for the msg and the reply */
+   memset (&msg, 0, sizeof (msg));
+   memset (&msg_reply, 0, sizeof (msg_reply));
+
+   /* Setup the message data to send to the server */
+   num = 2;                                                               
                                                      
/*********************** NUM 2 MEANS UPDATE OFFSET ***/
+   snprintf (msg.msg_data, 254, "client %d requesting reply.", getpid ());
+
+   printf ("client: msg_no: _IO_MAX + %d\n", num);
+   fflush (stdout);
+
+   ret = MsgSend (force_sensor_fd, &msg, sizeof (msg), msg_reply, 255);
+   if (ret == -1)
+     {
+ fprintf (stderr, "Unable to MsgSend() to server: %s\n",
+ strerror (errno));
+ return EXIT_FAILURE;
+     }
+   printf ("client: msg_reply:\n%s\n", msg_reply);
```

within servo on command, may be just after
```

292    if (strcmp(jname, "all") == 0 || strcmp(jname, "ALL") == 0){

```

as for
https://github.com/fkanehiro/hrpsys-base/tree/315.1.10/rtc/AbsoluteForceSensor

make sure that that RTC is crated and activated as 'rmfo' and properly 
connected, then try these service call
https://github.com/fkanehiro/hrpsys-base/blob/315.1.10/idl/AbsoluteForceSensorService.idl#L22

since this idl is not provided as latest hrpsys, so you need to provide 
within hironx package as we did for old ImpedanceControlller idl

--
◉ Kei Okada

On Mon, Apr 20, 2015 at 2:57 PM, Isaac I.Y. Saito <notifications@github.com> 
wrote:

> I say in advance that I'm sorry to ask many questions, but this is very 
> much complicated to me.
>
> may be I removed this code since when rtcd start, usually the hand 
> position is not good for calibration, 
>
> The code cited is in sample.c 
> <https://github.com/start-jsk/rtmros_hironx/blob/b499aad046f0e4818a798a033cbd9b576fde9605/hironx_ros_bridge/robot/nitta/sample.c#L70>
> .
>
> may be send '2' at 
> https://github.com/fkanehiro/hrpsys-base/blob/47cf3605ad409585aee23a9e7aa1e290471247fd/rtc/RobotHardware/robot.cpp#L293 
> when servo is on, is better.
>
> Could you tell me how to do so? 
>
> And are you referring to the right line?
>
> 292    if (strcmp(jname, "all") == 0 || strcmp(jname, "ALL") == 0){
> 293        bool ret = true;
> 294        for (int i=0; i<numJoints(); i++){
>
>  next step is to use following rtc, which takes care of this feature. it 
> does not require source compile, just to change startup script 
> https://github.com/fkanehiro/hrpsys-base/tree/315.1.10/rtc/AbsoluteForceSensor
>
> I have no idea how to do so.
> Isn't this line 
> <https://github.com/start-jsk/rtmros_hironx/commit/f551037aa697c1cf8b0faf1291e4c3565eba0059#diff-f9592097d70ddcac8ad91644f3717047R199> 
> for example doing what you meant??
>
> —
> Reply to this email directly or view it on GitHub 
> <https://github.com/start-jsk/rtmros_hironx/issues/357#issuecomment-94364517>
> .
>

130s, 2015-04-21:
Hope I understand better now.

> My thought was to put:
> :
> within servo on command, may be just after this line:

I've [made a change in `rtc/RobotHardware/robot.cpp`](https://github.com/130s/hrpsys-base/commit/ee2f800964a492df0d6392e762b4df06c10e8a54), without having opened a pull request yet. If this looks good I'll do so.

> since this idl is not provided as latest hrpsys, so you need to provide within hironx package as we did for old ImpedanceControlller idl

I have no idea how to create `_idl.py` for `AbsoluteForceSensor`, like [you did for ImpedanceControllerService_idl.py](https://github.com/start-jsk/rtmros_hironx/pull/351).

k-okada, 2015-04-22:
> I have no idea how to create _idl.py for AbsoluteForceSensor, like you 
did for ImpedanceControllerService_idl.py.

- just to roll back hrpsys-base and compile them (_idl.py is created from 
.idl)
- get ald AbsoluteForceSensor.idl and manually compile with omniidl (see 
makefile for option)
- copy from robot controller computer
- extract from old deb file

130s, 2015-12-04:
How-to is now summarized in wiki. http://wiki.ros.org/rtmros_nextage/Tutorials/Adding%20feedback%20control#Weight_offset

closed 2015-12-04 by 130s