start-jsk/rtmros_hironx #363: AttributeError: class ImpedanceControllerService has no attribute 'MODE_IMP'

130s, 2015-05-12:
Critical: this makes hironx.py ***unusable***.

hrpsys on QNX is 315.2.8.
```
$ ipython -i `rospack find hironx_ros_bridge`/scripts/acceptancetest_hironx.py -- --host nextage
Python 2.7.3 (default, Dec 18 2014, 19:10:20) 
Type "copyright", "credits" or "license" for more information.

IPython 0.12.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, *where)
    173             else:
    174                 filename = fname
--> 175             __builtin__.execfile(filename, *where)

/opt/ros/hydro/share/hironx_ros_bridge/scripts/acceptancetest_hironx.py in <module>()
     36 
     37 from hironx_ros_bridge.constant import Constant
---> 38 from hironx_ros_bridge.hironx_client import HIRONX
     39 from hironx_ros_bridge.ros_client import ROS_Client
     40 from hironx_ros_bridge.testutil.abst_acceptancetest import AbstAcceptanceTest

/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py in <module>()
     83                     pass
     84 
---> 85 class HIRONX(HrpsysConfigurator):
     86     '''
     87     @see: <a href = "https://github.com/fkanehiro/hrpsys-base/blob/master/" +

/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py in HIRONX()
    758                        reference_gain = 0.0,
    759                        manipulability_limit = 0.1,
--> 760                        controller_mode = OpenHRP.ImpedanceControllerService.MODE_IMP):
    761         self.ic_svc.setImpedanceControllerParam(
    762             arm,

AttributeError: class ImpedanceControllerService has no attribute 'MODE_IMP'

In [1]: 
Do you really want to exit ([y]/n)? 
marsryu@ayame:~$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py
Python 2.7.3 (default, Dec 18 2014, 19:10:20) 
Type "copyright", "credits" or "license" for more information.

IPython 0.12.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, *where)
    173             else:
    174                 filename = fname
--> 175             __builtin__.execfile(filename, *where)

/home/marsryu/catkin_ws/src/rtmros_nextage/nextage_ros_bridge/script/nextage.py in <module>()
     39 # This should come earlier than later import.

     40 # See http://code.google.com/p/rtm-ros-robotics/source/detail?r=6773

---> 41 from nextage_ros_bridge import nextage_client
     42 
     43 from hrpsys import rtm

/home/marsryu/catkin_ws/src/rtmros_nextage/nextage_ros_bridge/src/nextage_ros_bridge/nextage_client.py in <module>()
     35 # Author: Isaac Isao Saito

     36 
---> 37 from hironx_ros_bridge.hironx_client import HIRONX
     38 
     39 from nextage_ros_bridge.iros13_hands import Iros13Hands

/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py in <module>()
     83                     pass
     84 
---> 85 class HIRONX(HrpsysConfigurator):
     86     '''
     87     @see: <a href = "https://github.com/fkanehiro/hrpsys-base/blob/master/" +

/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py in HIRONX()
    758                        reference_gain = 0.0,
    759                        manipulability_limit = 0.1,
--> 760                        controller_mode = OpenHRP.ImpedanceControllerService.MODE_IMP):
    761         self.ic_svc.setImpedanceControllerParam(
    762             arm,

AttributeError: class ImpedanceControllerService has no attribute 'MODE_IMP'

$ rosversion hironx_ros_bridge
1.0.31
$ dpkg -p ros-hydro-hironx-ros-bridge |grep Ver
Version: 1.0.31-0precise-20150507-0951-+0000
```
FYI @RyuYamamoto

k-okada, 2015-05-12:
please do not edit log message, please copy and past all logs.

130s, 2015-05-12:
いや，コピーペーストして，どこも変えてないと思います

130s, 2015-05-12:
With nextage.py.
```
n130s@ayame:~$ ipython -i `rospack find nextage_ros_bridge`/script/nextage.py -- --host nextage
Python 2.7.3 (default, Dec 18 2014, 19:10:20) 
Type "copyright", "credits" or "license" for more information.

IPython 0.12.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, *where)
    173             else:
    174                 filename = fname
--> 175             __builtin__.execfile(filename, *where)

/opt/ros/hydro/share/nextage_ros_bridge/script/nextage.py in <module>()
     39 # This should come earlier than later import.

     40 # See http://code.google.com/p/rtm-ros-robotics/source/detail?r=6773

---> 41 from nextage_ros_bridge import nextage_client
     42 
     43 from hrpsys import rtm

/opt/ros/hydro/lib/python2.7/dist-packages/nextage_ros_bridge/nextage_client.py in <module>()
     35 # Author: Isaac Isao Saito

     36 
---> 37 from hironx_ros_bridge.hironx_client import HIRONX
     38 
     39 from nextage_ros_bridge.iros13_hands import Iros13Hands

/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py in <module>()
     83                     pass
     84 
---> 85 class HIRONX(HrpsysConfigurator):
     86     '''
     87     @see: <a href = "https://github.com/fkanehiro/hrpsys-base/blob/master/" +

/opt/ros/hydro/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.py in HIRONX()
    758                        reference_gain = 0.0,
    759                        manipulability_limit = 0.1,
--> 760                        controller_mode = OpenHRP.ImpedanceControllerService.MODE_IMP):
    761         self.ic_svc.setImpedanceControllerParam(
    762             arm,

AttributeError: class ImpedanceControllerService has no attribute 'MODE_IMP'

In [1]: 
```

130s, 2015-05-12:
The error above happens on Hydro machine. On my Indigo machine this error doesn't happen.
When this error happens, the hrpsys client class is not instantiated so checking its version by `robot.hrpsys_version` is not possible.

130s, 2015-05-12:
```
sudo apt-get install --reinstall ros-hydro-hironx* ros-hydro-nextage-*

$ more /etc/apt/sources.list.d/ros-latest.list
#deb http://packages.ros.org/ros/ubuntu precise main

# Shadow-fixed
deb http://packages.ros.org/ros-shadow-fixed/ubuntu precise main
```

Reinstall is no use for this Hydro machine.

130s, 2015-05-12:
```
sudo apt-get install --reinstall ros-hydro-hironx* ros-hydro-nextage-* ros-hydro-hrpsys*
```
This solved this. Thank you!

closed 2015-05-12 by 130s