start-jsk/rtmros_hironx #364: [rtmlaunch] [ERROR] Could not Connect ( /nextage:15005/RobotHardware0.rtc:q

130s, 2015-05-19:
Is this bad or not?
```
$ roslaunch nextage_ros_bridge nextage_ros_bridge_real.launch 
:
[rtmlaunch] Wait for  /nextage:15005/co.rtc:qCurrent   28 /30
[ INFO] [1432020070.034253965]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 45[Hz]
[ INFO] [1432020071.138539154]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 32[Hz]
[ INFO] [1432020072.174675614]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 45[Hz]
[ INFO] [1432020073.209227796]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 35[Hz]
[ INFO] [1432020074.278937111]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 35[Hz]
[rtmlaunch] Wait for  /nextage:15005/co.rtc:qCurrent   29 /30
[ INFO] [1432020075.280067594]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 52[Hz]
[rtmlaunch] [ERROR] Could not Connect ( /nextage:15005/RobotHardware0.rtc:q , /nextage:15005/co.rtc:qCurrent ):  No such object: /nextage:15005/co.rtc:qCurrent 
```
I'm seeing this error but so far the robot works perfectly. This line above happens periodically. 

```
$ dpkg -p ros-hydro-nextage-ros-bridge ros-hydro-hrpsys-ros-bridge ros-hydro-hrpsys ros-hydro-openhrp3 |grep Ver
Version: 0.6.1-0precise-20150311-1752-+0000
Version: 1.2.8-0precise-20150311-1659-+0000
Version: 315.3.0-1precise-20150311-1614-+0000
Version: 3.1.7-3precise-20150213-1403-+0000
```

130s, 2015-05-26:
@k-okada コメントが空のようです

k-okada, 2015-05-31:
please check hironx version, this should be fiexed in 1.0.31 ->

https://github.com/start-jsk/rtmros_hironx/commit/08e1fae2ad34fc6fe5d6db0f31cdc5550d202e80
--
◉ Kei Okada

130s, 2015-06-20:
With 1.0.31 I no longer see this issue. Thank you.

closed 2015-06-20 by 130s