start-jsk/rtmros_hironx #365: Sim hrpsys process keeps dying on Indigo

130s, 2015-06-18:
I'm working on a Trusty machines via TeamViewer (remote desktop service) where I see this issue.
```
roboworks@M14xR2:~$ rtmlaunch hironx_ros_bridge hironx_ros_bridge_simulation.launch 
[rtmlaunch] Start omniNames at port 15005 

Fri Jun 19 07:02:24 2015:

Read log file successfully.
:
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
auto-starting new master
process[master]: started with pid [12126]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b3b3e0a6-1605-11e5-9cc5-d4bed9726800
process[rosout-1]: started with pid [12139]
started core service [/rosout]
process[modelloader-2]: started with pid [12163]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys-3]: started with pid [12166]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12167]
IDL:omg.org/CORBA/TRANSIENT:1.0
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
process[hrpsys_py-4]: started with pid [12178]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12179]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[HrpsysSeqStateROSBridge-5]: started with pid [12185]
process[HrpsysJointTrajectoryBridge-6]: started with pid [12189]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12190]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys_state_publisher-7]: started with pid [12199]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12218]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys_ros_diagnostics-8]: started with pid [12221]
process[diagnostic_aggregator-9]: started with pid [12224]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12246]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys_profile-10]: started with pid [12258]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12259]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[sensor_ros_bridge_connect-11]: started with pid [12273]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12322]
IDL:omg.org/CORBA/TRANSIENT:1.0
/opt/ros/indigo/share/hrpsys_ros_bridge/scripts/diagnostics.py:75: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub = rospy.Publisher('diagnostics', DiagnosticArray)
process[rtmlaunch_hrpsys_ros_bridge-12]: started with pid [12335]
[modelloader-2] process has finished cleanly
[... 135 lines cut ...]
[hrpsys-3] killing on exit
[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[rtmlaunch] terminate omniNames at port 15005 
```
Versions:
```
$ dpkg -p ros-indigo-openrtm-aist ros-indigo-openhrp3 ros-indigo-hrpsys ros-indigo-hrpsys-ros-bridge ros-indigo-hironx-ros-bridge |grep Ver
Version: 1.1.0-26trusty-20150424-0428-+0000
Version: 3.1.8-0trusty-20150424-0521-+0000
Version: 315.4.0-0trusty-20150522-2045-+0000
Version: 1.2.12-0trusty-20150610-1738-+0000
Version: 1.0.31-0trusty-20150611-0259-+0000
```
`robot_model` uses source that includes https://github.com/ros/robot_model/pull/108

On another Trusty machine, however, the same simulation runs fine. I noticed that the s/w version is the same but the built date is slightly different.
```
$ dpkg -p ros-indigo-openrtm-aist|grep Ver
Version: 1.1.0-26trusty-20150514-090305-0700                                                                                       
```

130s, 2015-06-18:
##Rewritting the OP

I'm working on a Trusty machines via TeamViewer (remote desktop service) where I see this issue.
```
roboworks@M14xR2:~$ rtmlaunch hironx_ros_bridge hironx_ros_bridge_simulation.launch 
[rtmlaunch] Start omniNames at port 15005 

Fri Jun 19 07:02:24 2015:

Read log file successfully.
:
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
auto-starting new master
process[master]: started with pid [12126]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b3b3e0a6-1605-11e5-9cc5-d4bed9726800
process[rosout-1]: started with pid [12139]
started core service [/rosout]
process[modelloader-2]: started with pid [12163]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys-3]: started with pid [12166]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12167]
IDL:omg.org/CORBA/TRANSIENT:1.0
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
process[hrpsys_py-4]: started with pid [12178]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12179]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[HrpsysSeqStateROSBridge-5]: started with pid [12185]
process[HrpsysJointTrajectoryBridge-6]: started with pid [12189]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12190]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys_state_publisher-7]: started with pid [12199]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12218]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys_ros_diagnostics-8]: started with pid [12221]
process[diagnostic_aggregator-9]: started with pid [12224]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12246]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[hrpsys_profile-10]: started with pid [12258]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12259]
IDL:omg.org/CORBA/TRANSIENT:1.0
process[sensor_ros_bridge_connect-11]: started with pid [12273]
[modelloader-2] process has finished cleanly
log file: /home/roboworks/.ros/log/b3b3e0a6-1605-11e5-9cc5-d4bed9726800/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [12322]
IDL:omg.org/CORBA/TRANSIENT:1.0
/opt/ros/indigo/share/hrpsys_ros_bridge/scripts/diagnostics.py:75: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub = rospy.Publisher('diagnostics', DiagnosticArray)
[... 126 lines cut ...]
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[rtmlaunch] terminate omniNames at port 15005 
```

- Versions:

 ```
$ dpkg -p ros-indigo-openrtm-aist ros-indigo-openhrp3 ros-indigo-hrpsys ros-indigo-hrpsys-ros-bridge ros-indigo-hironx-ros-bridge |grep Ver
Version: 1.1.0-26trusty-20150424-0428-+0000
Version: 3.1.8-0trusty-20150424-0521-+0000
Version: 315.4.0-0trusty-20150522-2045-+0000
Version: 1.2.12-0trusty-20150610-1738-+0000
Version: 1.0.31-0trusty-20150611-0259-+0000
```
- `robot_model` uses source that includes https://github.com/ros/robot_model/pull/108
- On another Trusty machine, however, the same simulation runs fine. I noticed that the s/w version is the same but the built date is slightly different.

 ```
$ dpkg -p ros-indigo-openrtm-aist|grep Ver
Version: 1.1.0-26trusty-20150514-090305-0700
```

130s, 2015-06-18:
Found `omniNames` was running but stopping it takes no effect.

```
$ sudo service omniorb4-nameserver stop
[sudo] password for tork: 
 * Stopping omniORB name server omniNames
```

130s, 2015-06-19:
After restarting the OS the issue is gone.

closed 2015-06-19 by 130s

k-okada, 2015-06-21:
note : sudo service .. will stop nameserver with default port 2809
rtmros uses nameserver with custom port 15005, which is started on 
rtmrlaumch (
https://github.com/start-jsk/rtmros_common/blob/master/openrtm_tools/scripts/rtmlaunch#L31
)