start-jsk/rtmros_hironx #366: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

130s, 2015-06-19:
On Indigo, RViz fails to load 3d model. For `robot_model` is built from source to include https://github.com/ros/robot_model/pull/108.
```
$ roslaunch hironx_moveit_config moveit_planning_execution.launch 
... logging to /home/n130s/.ros/log/964b6b46-1660-11e5-9e3a-f816542d218e/roslaunch-tork-kudu1-4118.log
:
core service [/rosout] found
process[move_group-1]: started with pid [4201]
process[rviz_tork_kudu1_4118_6007383188480671338-2]: started with pid [4274]
[ INFO] [1434704122.968404649]: rviz version 1.11.7
[ INFO] [1434704122.970581116]: compiled against OGRE version 1.8.1 (Byatis)
[ERROR] [1434704123.070017339, 140.604999999]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

[ERROR] [1434704123.083055133, 140.614999999]: COLLADA error: Failed to load XML document from memory

[ERROR] [1434704123.084098711, 140.614999999]: Unable to parse URDF
[ERROR] [1434704123.084424260, 140.614999999]: Robot model not loaded
[ERROR] [1434704123.104024317, 140.634999999]: Planning scene not configured
[ INFO] [1434704123.224929142]: Stereo is NOT SUPPORTED
[ INFO] [1434704123.225028724]: OpenGl version: 3 (GLSL 1.3).
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
[move_group-1] process has died [pid 4201, exit code -6, cmd /opt/ros/indigo/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/n130s/.ros/log/964b6b46-1660-11e5-9e3a-f816542d218e/move_group-1.log].
log file: /home/n130s/.ros/log/964b6b46-1660-11e5-9e3a-f816542d218e/move_group-1*.log
[ERROR] [1434704128.627028290, 146.159999999]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

[ERROR] [1434704128.637774920, 146.174999999]: COLLADA error: Failed to load XML document from memory

[ERROR] [1434704128.639009001, 146.174999999]: Unable to parse URDF
[ERROR] [1434704128.639340338, 146.174999999]: Robot model not loaded
```

k-okada, 2015-06-19:
> source to include ros/robot_model#108

you must build rviz too to use library include in robot_model. I think your 
rviz using old robot_model under /opt

130s, 2015-07-22:
> you must build rviz too to use library include in robot_model. I think your rviz using old robot_model under /opt

hmm, looks like I'm doing the right but still get the same error. `robot_model` is the latest indigo branch, while rviz is on the released tag, all built by catkin_tools.

```
/home/rospasta/cws_hironxo/src/start-jsk/rtmros_hironx/hironx_ros_bridge
rospasta@tork-kudu1:~//cws_hironxo$ roscd rviz
rospasta@tork-kudu1:~/cws_hironxo/src/ros-visualization/rviz$ git branch
* 1.11.7
  fix-879
  fix-879_2
  hydro-devel
  indigo-devel
rospasta@tork-kudu1:~/cws_hironxo/src/ros-visualization/rviz$ git log
commit 11fcdadbbcc4c9d38a0bd4d580be6f0b49cbbc47
Author: William Woodall <william@osrfoundation.org>
Date:   Mon Mar 2 13:09:25 2015 -0800

    1.11.7

:
rospasta@tork-kudu1:~/cws_hironxo/src/ros-visualization/rviz$ roscd collada_urdf
rospasta@tork-kudu1:~/cws_hironxo/src/ros/robot_model/collada_urdf$ git branch
* indigo-devel
rospasta@tork-kudu1:~/cws_hironxo/src/ros/robot_model/collada_urdf$ git log
commit 1a1f0db396901e66689d3a8949b9bcee3653e102
Merge: 6534424 c4073db
Author: Ioan A Sucan <isucan@users.noreply.github.com>
Date:   Wed Jul 15 10:00:27 2015 -0700

    Merge pull request #113 from k-okada/remove_pcrecpp_stuff
    
    revert #108, #106, #105
:
rospasta@tork-kudu1:~/cws_hironxo/src/ros-visualization/rviz$ roslaunch hironx_moveit_config moveit_planning_execution.launch 
:
process[move_group-1]: started with pid [14713]
process[rviz_tork_kudu1_14695_9030492380499235487-2]: started with pid [14746]
[ INFO] [1437526362.154180076]: rviz version 1.11.7
[ INFO] [1437526362.154276768]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1437526362.434720173, 19.115000000]: Stereo is NOT SUPPORTED
[ INFO] [1437526362.434951439, 19.115000000]: OpenGl version: 3 (GLSL 1.3).
[ERROR] [1437526368.598094948, 25.294999999]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

[ERROR] [1437526368.608727096, 25.304999999]: COLLADA error: Failed to load XML document from memory

[ERROR] [1437526368.610306110, 25.309999999]: Unable to parse URDF
[ERROR] [1437526368.610780945, 25.309999999]: Robot model not loaded
:
$ ps -ef|grep rviz
rospasta    16571 16520 42 18:06 ?        00:00:03 /home/rospasta/cws_hironxo/devel/lib/rviz/rviz -d /home/rospasta/cws_hironxo/src/start-jsk/rtmros_hironx/hironx_moveit_config/launch/moveit_viewnatto.rviz __name:=rviz_tork_kudu1_16520_4824946678805607619 __log:=/home/rospasta/.ros/log/c38a1dd4-300d-11e5-ae74-f816542d218e/rviz_tork_kudu1_16520_4824946678805607619-2.log
rospasta    16754  5351  0 18:06 pts/16   00:00:00 grep --color=auto rviz
```

k-okada, 2015-07-22:
situation has changed, 
1a1f0db396901e66689d3a8949b9bcee3653e102 should not be merged until newer version of collada-dom is releaesd, 
if you  latest code as of today, you have to use newer version of collada-dom http://lists.ros.org/pipermail/ros-release/2015-June/004650.html

130s, 2015-07-23:
Temporary workaround is found. 

```
$ curl -sL https://raw.githubusercontent.com/tork-a/88_tools/master/collada_dom/create_dom.sh | bash
```

This script downloads source of `collada_dom` 2.4.3, apply the necessary patch, builds and install all locally.

Once DEB version of `collada_dom` becomes available, this won't be needed. Until then, this ticket remains open.

Credit goes to @k-okada

emijah, 2015-08-07:
@130s @k-okada Tried ROS Indigo on a clean install of 14.04 today and found that both the simulation and the python interface works.

130s, 2015-08-13:
@emijah thank you for the update. Did you run Hironx software?

NEXTAGE software should have been up running without the issue in this ticket. This ticket is about Hironx software.

130s, 2015-10-03:
https://github.com/rdiankov/collada-dom/issues/16 is done thanks to @k-okada, so this issue should be resolved.

closed 2015-10-03 by 130s