start-jsk/rtmros_hironx #367: EOL hydro-branch

130s, 2015-06-20:
Considering that we move on to ROS Indigo and will likely stick to it for another few years as a stable distro, I just created a [indigo-devel](https://github.com/start-jsk/rtmros_hironx/tree/indigo-devel) branch, and made it as a default.

Knowing that there are groups of [users who'll stay on Hydro for awhile](https://groups.google.com/forum/#!topic/rtm-ros-robotics/qrDWvnjxrV0), we'll keep backporting applicable changes to `hydro-devel` branch as well.

Question is when we'll stop synchronizing (which requires testing) hydro branch. I think until the end of September is doable.

Ref. #202

k-okada, 2015-06-20:
Is there any pros. on creating indigo branch? as long as we do not have any
distro specific source code, we do not need that one and continue working
on master until the end of rosdistro sync.

130s, 2015-07-01:
I can think of at minimum 2 advantages to have distro specific branches:
- To better indicate that the old ROS distro may not get maintenance, including testing on the robot hardware.
- Since Indigo will stay longer than any other ROS distro ever, I expect in the future there will be features that rely on functions available only in newer ROS distro.

I thought it would be less complexing if we created a distro specific branch now than we do so later, but I can agree to create specific branch when it becomes necessary (because to the best we won't need to create it), as long as users are ok with the idea that testing with robot hardware in Hydro version will be limited.

130s, 2015-07-01:
Now `hydro-devel` is set as default.

k-okada, 2015-07-05:
1) i think current situation is very special that hydro and indigo is very 
separated, one works on 12.04 and another requires 14.04, but if you think 
about indigo-jade, they are very similar, both will used on 14.04, so 
basically we should focus on ubuntu version, not ros version
2) I understand that we'd like to create distro specific branch if there 
there difference between distros, but most of "stable" package did not 
change API between ros distro, they keep old API at least one distro, so 
basically two successive distro has compatibility,  and our software should 
depends on such package.
3) I think we should create tag (or maybe branch?) for hydro when ros build 
firm stops, so we can keep, but maybe bumping major version in such 
situation may sufficient.

130s, 2015-10-21:
> basically we should focus on ubuntu version, not ros version

This makes sense. I've made `indigo-devel` as the default.

Also I just made [a tag 1.1.0](made https://github.com/start-jsk/rtmros_hironx/releases/tag/1.1.0). 

For Hydro for hironx,

- 1.0.37 is the last release.
- Source code will be kept available at `hydro-devel` branch, but no newer commits will go in there unless requested.

closed 2015-10-21 by 130s

130s, 2015-10-21:
DEB release is in flight now. https://github.com/ros/rosdistro/pull/9687