start-jsk/rtmros_hironx #369: [Call for volunteer] Robot page on roswiki

130s, 2015-07-01:
By following the defacto standard on the ["Robots" page on ROS wiki](http://wiki.ros.org/Robots), I finally created HiroNXO page. 

http://wiki.ros.org/Robots/HiroNXO

Please have a look and add any necessary change.

There's no "rule" per se but here's what I stick to:
- Focus is on software, but decent info on hardware is probably appreciated by the readers
- List capabilities and ROS-formatted name of modules so that readers on that site easily have ideas of what features available in ROS can be used with this robot
- No too commercial promotion (respect that ROS wiki is opensource communication platform)

One of the things especially appreciated is the feature-by-feature images.

130s, 2015-07-01:
I close this ticket but please feel free to make/suggest changes.

closed 2015-07-01 by 130s

k-okada, 2015-07-04:
nice!!

emijah, 2015-07-08:
Can't think of anything to add at this point...