start-jsk/rtmros_hironx #371: RViz repeats to crash and restart

130s, 2015-07-15:
[A user reported](https://groups.google.com/forum/#!topic/rtm-ros-robotics/zjrg8GJJip4) the issue both on simulation and real robot. 

Updating DEB packages solved as he said. I assume the versions are:

Old: 1.0.27-0precise-20141206-2255-+0000
New: 1.0.31

```
rospasta@robonoodle:~$ roslaunch hironx_moveit_config moveit_planning_execution.launch 
... logging to /home/ais/.ros/log/d1755138-2389-11e5-ae60-00259093db68/roslaunch-yodo-12243.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yodo:36832/

SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution
 * /move_group/capabilities
 * /move_group/controller_list
 * /move_group/controller_manager_name
 * /move_group/controller_manager_ns
 * /move_group/jiggle_fraction
 * /move_group/left_arm/kinematics_solver
 * /move_group/left_arm/kinematics_solver_attempts
 * /move_group/left_arm/kinematics_solver_search_resolution
 * /move_group/left_arm/kinematics_solver_timeout
 * /move_group/left_arm/longest_valid_segment_fraction
 * /move_group/left_arm/planner_configs
 * /move_group/left_arm/projection_evaluator
 * /move_group/left_gripper/kinematics_solver
 * /move_group/left_gripper/kinematics_solver_attempts
 * /move_group/left_gripper/kinematics_solver_search_resolution
 * /move_group/left_gripper/kinematics_solver_timeout
 * /move_group/left_gripper/longest_valid_segment_fraction
 * /move_group/left_gripper/planner_configs
 * /move_group/left_gripper/projection_evaluator
 * /move_group/max_safe_path_cost
 * /move_group/moveit_controller_manager
 * /move_group/moveit_manage_controllers
 * /move_group/planner_configs/BKPIECEkConfigDefault/type
 * /move_group/planner_configs/ESTkConfigDefault/type
 * /move_group/planner_configs/KPIECEkConfigDefault/type
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type
 * /move_group/planner_configs/LazyRRTkConfigDefault/type
 * /move_group/planner_configs/RRTConnectkConfigDefault/type
 * /move_group/planner_configs/RRTStarkConfigDefault/type
 * /move_group/planner_configs/RRTkConfigDefault/type
 * /move_group/planner_configs/SBLkConfigDefault/type
 * /move_group/planning_plugin
 * /move_group/planning_scene_monitor/publish_geometry_updates
 * /move_group/planning_scene_monitor/publish_planning_scene
 * /move_group/planning_scene_monitor/publish_state_updates
 * /move_group/planning_scene_monitor/publish_transforms_updates
 * /move_group/request_adapters
 * /move_group/right_arm/kinematics_solver
 * /move_group/right_arm/kinematics_solver_attempts
 * /move_group/right_arm/kinematics_solver_search_resolution
 * /move_group/right_arm/kinematics_solver_timeout
 * /move_group/right_arm/longest_valid_segment_fraction
 * /move_group/right_arm/planner_configs
 * /move_group/right_arm/projection_evaluator
 * /move_group/right_gripper/kinematics_solver
 * /move_group/right_gripper/kinematics_solver_attempts
 * /move_group/right_gripper/kinematics_solver_search_resolution
 * /move_group/right_gripper/kinematics_solver_timeout
 * /move_group/right_gripper/longest_valid_segment_fraction
[... 248 lines cut ...]
[ INFO] [1436151797.759156389]: Loading robot model 'HiroNX'...
[ERROR] [1436151797.814806250]: Group 'right_gripper' is not a chain
[ERROR] [1436151797.814869106]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'right_gripper'
[ERROR] [1436151797.815515712]: Kinematics solver could not be instantiated for joint group right_gripper.
[ INFO] [1436151798.075950425]: Loading robot model 'HiroNX'...
[ERROR] [1436151798.120713130]: Group 'left_gripper' is not a chain
[ERROR] [1436151798.120760395]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'left_gripper'
[ERROR] [1436151798.120920780]: Kinematics solver could not be instantiated for joint group left_gripper.
[ INFO] [1436151798.260502058]: Starting scene monitor
[ INFO] [1436151798.262965129]: Listening to '/move_group/monitored_planning_scene'
[ WARN] [1436151798.267279314]: No transform available between frame 'base_footprint' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1436151798.267379121]: No transform available between frame 'odom' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'odom' because they are not part of the same tree.Tf has two or more unconnected trees.)
[ WARN] [1436151798.267444951]: No transform available between frame 'BODY' and planning frame '/WAIST' (Could not find a connection between 'WAIST' and 'BODY' because they are not part of the same tree.Tf has two or more unconnected trees.)
[rviz_yodo_12243_6992361472571982871-2] process has died [pid 13624, exit code -11, cmd /opt/ros/hydro/lib/rviz/rviz -d /opt/ros/hydro/share/hironx_moveit_config/launch/moveit_viewnatto.rviz __name:=rviz_yodo_12243_6992361472571982871 __log:=/home/ais/.ros/log/d1755138-2389-11e5-ae60-00259093db68/rviz_yodo_12243_6992361472571982871-2.log].
log file: /home/ais/.ros/log/d1755138-2389-11e5-ae60-00259093db68/rviz_yodo_12243_6992361472571982871-2*.log
respawning...
[rviz_yodo_12243_6992361472571982871-2] restarting process
process[rviz_yodo_12243_6992361472571982871-2]: started with pid [14116]
[ INFO] [1436151798.731904066]: rviz version 1.10.18
[ INFO] [1436151798.732095740]: compiled against OGRE version 1.7.4 (Cthugha)
[ INFO] [1436151799.319660624]: Stereo is NOT SUPPORTED
[ INFO] [1436151799.319788842]: OpenGl version: 4.2 (GLSL 4.2).
[ WARN] [1436151801.161056259]: The complete state of the robot is not yet known.  Missing LHAND_JOINT0, LHAND_JOINT1, LHAND_JOINT2, LHAND_JOINT3, RHAND_JOINT0, RHAND_JOINT1, RHAND_JOINT2, RHAND_JOINT3
^C[rviz_yodo_12243_6992361472571982871-2] killing on exit
```

closed 2015-07-15 by 130s

SantoshBanisetty, 2017-05-22:
@130s I am having a similar issue with pr2. Do you happen to know a solution for the same? Thanks.

130s, 2017-05-22:
@SantoshBanisetty Are you using Hironx / NEXTAGE? 
- If so we *could* continue discussion on this ticket.
- If not, this ticket is not a right place for your question. Go ask on http://answers.ros.org/

Either way, you need to provide more info in general (e.g. full output of `roslaunch` etc. See http://wiki.ros.org/Support for the type of info you should provide when asking questions).