start-jsk/rtmros_hironx #375: velocity & position limit over print messages never stop

130s, 2015-07-22:
Even while no command is passed to the robot on the simulator nor the robot doesn't seem moving, this print floods the simulator's terminal. Tried some commands (robot.goInitial, goOffpose) and seemed to work fine.
 
Is this print bad, or can we just ignore (if so is there any way to suppress printing)?

```
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
position limit over (0), qRef=-8.75e-07, llimit =0, ulimit =0, servo_state = ON, prev_angle = 0
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
position limit over (0), qRef=-8.75e-07, llimit =0, ulimit =0, servo_state = ON, prev_angle = 0
[ INFO] [1437524997.573169080, 4.739999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 208[Hz] (exec 1762 Hz, dropped 208)
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
position limit over (0), qRef=-8.75e-07, llimit =0, ulimit =0, servo_state = ON, prev_angle = 0
[INFO] [WallTime: 1437524997.839578] [5.005000] check 4 collision status, 0.799201 Hz
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
position limit over (0), qRef=-8.75e-07, llimit =0, ulimit =0, servo_state = ON, prev_angle = 0
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
position limit over (0), qRef=-8.75e-07, llimit =0, ulimit =0, servo_state = ON, prev_angle = 0
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
position limit over (0), qRef=-8.75e-07, llimit =0, ulimit =0, servo_state = ON, prev_angle = 0
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
position limit over (0), qRef=-8.75e-07, llimit =0, ulimit =0, servo_state = ON, prev_angle = 0
[ INFO] [1437524998.576652564, 5.744999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1780 Hz, dropped 201)
velocity limit over (0), qvel=0, lvlimit =0.000175, uvlimit =-0.000175, servo_state = ON
```
Versions:

```
$ dpkg -p ros-indigo-hrpsys|grep Ver
Version: 315.6.0-0trusty-20150711-0953-+0000
$ rospack find hironx_ros_bridge 
/home/rospasta/cws_hironxo/src/start-jsk/rtmros_hironx/hironx_ros_bridge
$ rosversion hironx_ros_bridge
1.0.32
```

More elaborate log:
```
$ rtmlaunch hironx_ros_bridge hironx_ros_bridge_simulation.launch 
[rtmlaunch] Start omniNames at port 15005 

Tue Jul 21 17:29:51 2015:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000d0000003139322e3136382e312e363500009d3a0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000ffe3ae5501002ee3
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/rospasta/.ros/log/c4150a66-3008-11e5-b11f-f816542d218e/roslaunch-tork-kudu1-11997.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tork-kudu1:34513/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC']
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
[... 531 lines cut ...]
[hrpsys_profile-10] killing on exit
[diagnostic_aggregator-9] killing on exit
[hrpsys_ros_diagnostics-8] killing on exit
[HrpsysJointTrajectoryBridge-6] killing on exit
[HrpsysSeqStateROSBridge-5] killing on exit
[hrpsys-3] killing on exit
[modelloader-2] killing on exit
[ INFO] [1437524999.366628009, 6.534999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (larm
[ INFO] [1437524999.370210140, 6.534999999]: [HrpsysSeqStateROSBridge] @onFinalize : HrpsysSeqStateROSBridge0
terminate called after throwing an instance of 'CORBA::COMM_FAILURE'
[ INFO] [1437524999.376690860, 6.534999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rarm
terminate called after throwing an instance of 'CORBA::OBJECT_NOT_EXIST'
[ INFO] [1437524999.388196772, 6.534999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (head
[ INFO] [1437524999.403009578, 6.534999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (torso
[ INFO] [1437524999.410107335, 6.534999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (larm_torso
[ INFO] [1437524999.416481788, 6.534999999]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rarm_torso
terminate called after throwing an instance of 'CORBA::COMM_FAILURE'
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[rtmlaunch] terminate omniNames at port 15005 
rospasta@tork-kudu1:~/cws_hironxo$ 
```

130s, 2015-07-22:
The log in the previous post was on Indigo. 

Now tested on Hydro, this print didn't occur. But then only upgrading `hrpsys` DEB from 315.4.0 to 315.6.0, the same thing happens.

```
$ dpkg -p ros-hydro-hrpsys|grep Ver
Version: 315.6.0-0precise-20150711-0856-+0000
```

130s, 2015-07-22:
Btw on Indigo, I haven't fixed #366 yet (in fact I happened to find this issue while I was working on #366). So in the log above we still saw the same error.
```
[ERROR] [1437524993.297885948]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

process[diagnostic_aggregator-9]: started with pid [12174]
[ERROR] [1437524993.310043759]: COLLADA error: Failed to load XML document from memory
```

k-okada, 2015-07-22:
That should be resolved in 
https://github.com/fkanehiro/hrpsys-base/pull/726
--
◉ Kei Okada

130s, 2015-08-14:
I see, we'll wait for the next version of hrpsys https://github.com/fkanehiro/hrpsys-base/issues/757.

closed 2015-08-14 by 130s