start-jsk/rtmros_hironx #381: travis said ok, but the test was failing

k-okada, 2015-08-11:
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/74030660
```
Loading body customizer "/opt/ros/hydro/share/OpenHRP-3.1/customizer/libSampleCustomizer.so" for springJoint
testtest_hironx_controller ... ok
testtest_hironx_fullbody ... ok
testtest_hironx_ik ... ok
testtest_hironx_ik_noinit ... ok
testtest_hironx_limb ... ok
testtest_hironx_target ... ok

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
[1;31m * RESULT: FAIL[0m
 * TESTS: 0
[1;31m * ERRORS: 6[0m
 * FAILURES: 0
```
```
testtest_hironx_ros_bridge_send ... ok

[ROSTEST WARNINGS]--------------------------------------------------------------

 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtactivate

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
[1;31m * RESULT: FAIL[0m
 * TESTS: 0
[1;31m * ERRORS: 1[0m
 * FAILURES: 0
```
```
testtest_hironx_ros_bridge_controller ... ok

[ROSTEST WARNINGS]--------------------------------------------------------------

 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtactivate

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
[1;31m * RESULT: FAIL[0m
 * TESTS: 0
[1;31m * ERRORS: 1[0m
 * FAILURES: 0
```
```
 * unrecognized tag rtconnect
 * unrecognized tag rtactivate

[ROSTEST]-----------------------------------------------------------------------

[hironx_ros_bridge.rosunit-joint_states_test/test_hz][FAILURE]------------------
no messages before timeout
  File "/usr/lib/python2.7/unittest/case.py", line 327, in run
    testMethod()
  File "/opt/ros/hydro/share/rostest/nodes/hztest/hztest", line 115, in test_hz
    self._test_hz(hz, hzerror, topic, test_duration, wait_time)
  File "/opt/ros/hydro/share/rostest/nodes/hztest/hztest", line 150, in _test_hz
    self.assert_(self.message_received, "no messages before timeout")
  File "/usr/lib/python2.7/unittest/case.py", line 420, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

SUMMARY
[1;31m * RESULT: FAIL[0m
 * TESTS: 1
 * ERRORS: 0
[1;31m * FAILURES: 1[0m
```

130s, 2015-11-11:
I've looked at all the variants of 3 PRs recently merged, and I don't see a failure in any of them. For now I think it's hard to track down this problem.
Please reopen if we need to look into more details.

https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90558278
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90558279
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90558281
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90566195
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90566196
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90566197
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90576911
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90576912
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90576913

k-okada, 2015-11-12:
https://travis-ci.org/start-jsk/rtmros_hironx/jobs/90588783
```
+rostest /home/travis/catkin_ws/install_isolated/share/hironx_ros_bridge/test/test-hironx-ros-bridge-send.test
... logging to /home/travis/.ros/log/rostest-testing-gce-7954619f-d4e8-4c27-8bee-57eff1ed45f7-24494.log
[ROSUNIT] Outputting test results to /home/travis/.ros/test_results/hironx_ros_bridge/rostest-test_test-hironx-ros-bridge-send.xml

Wed Nov 11 20:48:13 2015:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000c00000031302e3234302e302e353200f90a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000008da9435601005fd3
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
ready
libdc1394 error: Failed to initialize libdc1394
loading file:///home/travis/catkin_ws/install_isolated/share/hironx_ros_bridge/models/kawada-hironx.dae
Collada Warning: fill asset info
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: joint HEAD_JOINT1 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint HEAD_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
loading /home/travis/catkin_ws/install_isolated/share/hironx_ros_bridge/models/kawada-hironx.dae
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: joint RHAND_JOINT1 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint RHAND_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: joint RHAND_JOINT3 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint RHAND_JOINT2 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint RARM_JOINT5 has axis: 1.000000 0.000000 0.000000
Collada Warning: joint RARM_JOINT4 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT3 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint RARM_JOINT2 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT1 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
Collada Warning: generate normals from vertices
Collada Warning: joint LHAND_JOINT1 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint LHAND_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: joint LHAND_JOINT3 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint LHAND_JOINT2 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint LARM_JOINT5 has axis: 1.000000 0.000000 0.000000
Collada Warning: joint LARM_JOINT4 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint LARM_JOINT3 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint LARM_JOINT2 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint LARM_JOINT1 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint LARM_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint CHEST_JOINT0 has axis: 0.000000 0.000000 1.000000
libdc1394 error: Failed to initialize libdc1394
[... 2083 lines cut ...]
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtactivate
 * unrecognized tag rtconnect
 * unrecognized tag rtconnect
 * unrecognized tag rtactivate

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
m * RESULT: FAILm
 * TESTS: 0
m * ERRORS: 1m
 * FAILURES: 0

rostest log file is in /home/travis/.ros/log/rostest-testing-gce-7954619f-d4e8-4c27-8bee-57eff1ed45f7-24494.log
+export TMP_EXIT_STATUS=1
+TMP_EXIT_STATUS=1
+'[' 1 '!=' 0 ']'
+export EXIT_STATUS=1
+EXIT_STATUS=1
++basename /home/travis/catkin_ws/install_isolated/share/hironx_ros_bridge/test/test-hironx-r

130s, 2015-11-12:
Ok, I also found [this phenomenon](https://travis-ci.org/tork-a/rtmros_nextage/jobs/90801599#L7371) (in a different repository though) in [a recent PR](https://github.com/tork-a/rtmros_nextage/pull/209).