start-jsk/rtmros_hironx #392: cannot connect to Hiro(real robot)

KodaiNakamura, 2015-08-28:
I'm trying to run hironx_ros_bridge and connect to Hiro, but I got following errors.

My enviroment is Ubuntu14.04.3 and indigo.

```
~ ❯❯❯ roslaunch hironx_ros_bridge hironx_ros_bridge_real.launch nameserver:=hiro
... logging to /home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/roslaunch-kodai-MacBookPro-11702.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kodai-MacBookPro:34598/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /collision_state/comp_name: CollisionDetector0
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC']
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /robot_pose_ekf/debug: False
 * /robot_pose_ekf/freq: 30.0
 * /robot_pose_ekf/imu_used: False
 * /robot_pose_ekf/odom_used: True
 * /robot_pose_ekf/output_frame: odom
 * /robot_pose_ekf/self_diagnose: False
 * /robot_pose_ekf/sensor_timeout: 1.0
 * /robot_pose_ekf/vo_used: False
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    CollisionDetectorComp (hrpsys/CollisionDetectorComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    RobotHardwareServiceROSBridge (hrpsys_ros_bridge/RobotHardwareServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
    ServoControllerServiceROSBridge (hrpsys_ros_bridge/ServoControllerServiceROSBridgeComp)
    StateHolderServiceROSBridge (hrpsys_ros_bridge/StateHolderServiceROSBridgeComp)
    base_footprint_rootlink (tf/static_transform_publisher)
    collision_state (hrpsys_ros_bridge/collision_state.py)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    hrpsys_profile (hrpsys_ros_bridge/hrpsys_profile.py)
    hrpsys_py (hironx_ros_bridge/hironx.py)
    hrpsys_ros_diagnostics (hrpsys_ros_bridge/diagnostics.py)
    hrpsys_state_publisher (robot_state_publisher/state_publisher)
    robot_pose_ekf (robot_pose_ekf/robot_pose_ekf)
    rtmlaunch_collision_detector (openrtm_tools/rtmlaunch.py)
    rtmlaunch_hrpsys_ros_bridge (openrtm_tools/rtmlaunch.py)
    sensor_ros_bridge_connect (hrpsys_ros_bridge/sensor_ros_bridge_connect.py)

WARNING: unrecognized tag rtconnect
[... 118 lines cut ...]
    self._parse_server(address, orb, filter)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/nameserver.py", line 78, in _parse_server
    root_context = self._connect_to_naming_service(address)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rtctree/nameserver.py", line 93, in _connect_to_naming_service
    raise InvalidServiceError(address)
rtctree.exceptions.InvalidServiceError: Invalid CORBA naming service: hiro:15005
[CollisionDetectorComp-19] process has died [pid 12097, exit code -6, cmd /opt/ros/indigo/lib/hrpsys/CollisionDetectorComp -o corba.nameservers:hiro:15005 -o naming.formats:%n.rtc -o exec_cxt.periodic.type:PeriodicExecutionContext -o exec_cxt.periodic.rate:2000 -o logger.file_name:/tmp/collision_detector%p.log -o example.CollisionDetector.config_file:/opt/ros/indigo/share/hironx_ros_bridge/conf/hironx_realrobot_fixedpath.conf __name:=CollisionDetectorComp __log:=/home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/CollisionDetectorComp-19.log].
log file: /home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/CollisionDetectorComp-19*.log
[rtmlaunch_collision_detector-20] process has died [pid 12127, exit code 1, cmd /opt/ros/indigo/share/openrtm_tools/scripts/rtmlaunch.py /opt/ros/indigo/share/hrpsys_ros_bridge/launch/collision_detector.launch __name:=rtmlaunch_collision_detector __log:=/home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/rtmlaunch_collision_detector-20.log].
log file: /home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/rtmlaunch_collision_detector-20*.log
/opt/ros/indigo/share/hrpsys_ros_bridge/scripts/collision_state.py:181: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub_diagnostics = rospy.Publisher('diagnostics', DiagnosticArray)
/opt/ros/indigo/share/hrpsys_ros_bridge/scripts/collision_state.py:182: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  pub_collision = rospy.Publisher('collision_detector_marker_array', MarkerArray)
configuration ORB with hiro:15005
[ERROR] Connection Failed with the Nameserver (hostname=hiro port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
Exception in thread Thread-3 (most likely raised during interpreter shutdown):[collision_state-21] process has died [pid 12128, exit code 1, cmd /opt/ros/indigo/share/hrpsys_ros_bridge/scripts/collision_state.py /opt/jsk/etc/HIRONX/model/main.wrl -ORBInitRef NameService=corbaloc:iiop:hiro:15005/NameService __name:=collision_state __log:=/home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/collision_state-21.log].
log file: /home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/collision_state-21*.log
[HrpsysJointTrajectoryBridge-4] process has finished cleanly
log file: /home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/HrpsysJointTrajectoryBridge-4*.log
[HrpsysSeqStateROSBridge-3] process has died [pid 11745, exit code -11, cmd /opt/ros/indigo/lib/hrpsys_ros_bridge/HrpsysSeqStateROSBridge -o corba.nameservers:hiro:15005 -o naming.formats:%n.rtc -o exec_cxt.periodic.type:PeriodicExecutionContext -o exec_cxt.periodic.rate:2000 -o logger.file_name:/tmp/rtc%p.log -o model:/opt/jsk/etc/HIRONX/model/main.wrl -o example.HrpsysSeqStateROSBridge.config_file: __name:=HrpsysSeqStateROSBridge __log:=/home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/HrpsysSeqStateROSBridge-3.log].
log file: /home/kodai/.ros/log/ea99d36a-4d63-11e5-b419-00904c0df43e/HrpsysSeqStateROSBridge-3*.log
```

130s, 2015-08-29:
Can you tell us the output of the following command?
```
rosrun rtshell rtls hiro:15005/
```

Also appreciable is to send the tarball of the following command to support@opensource-robotics.tokyo.jp (no body text is fine; Just refer to this ticket). Thank you.
```
curl -sL https://raw.githubusercontent.com/start-jsk/rtmros_common/master/hrpsys_tools/scripts/rtmroswtf.sh | bash -
```

KodaiNakamura, 2015-08-29:
thank for your reply, but I could resolve the problem.

I just forgot to set hiro's host name on /etc/hosts.

130s, 2015-08-29:
:)

closed 2015-08-29 by 130s