start-jsk/rtmros_hironx #419: COLLADA error: Failed to load XML document from memory

youtalk, 2015-12-20:
When I use the latest release package of `ros-indigo-rtmros-hironx`, rtmlaunch outputs a COLLADA error and no TF is published. 
It seems to be a bug of collada parser but I could not solve it. Do you have any idea?

~~~sh
% rtmlaunch hironx_ros_bridge hironx_ros_bridge_simulation.launch
Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
...
Collada Warning: joint RARM_JOINT5 has axis: 1.000000 0.000000 0.000000
Collada Warning: joint RARM_JOINT4 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT3 has axis: 0.000000 0.000000 1.000000
Collada Warning: joint RARM_JOINT2 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT1 has axis: 0.000000 1.000000 0.000000
Collada Warning: joint RARM_JOINT0 has axis: 0.000000 0.000000 1.000000
Collada Warning: generate normals from vertices
[ERROR] [1450595456.284363129]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
[ERROR] [1450595456.328304880]: COLLADA error: Failed to load XML document from memory

Collada Warning: generate normals from vertices
Collada Warning: generate normals from vertices
Collada Warning: joint LHAND_JOINT1 has axis: 0.000000 0.000000 1.000000
...
~~~

In case of the hydro release package, there is no COLLADA errror.

k-okada, 2015-12-21:
please check if you use collada 2.4.4 and if not, please download from
then, http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/c/collada-dom2.4-dp/

you need following 3 files

collada-dom2.4-dp-base_2.4.4.0-ubuntu1~trusty1_amd64.deb (1.3 MiB)
collada-dom2.4-dp-dev_2.4.4.0-ubuntu1~trusty1_amd64.deb (201.2 KiB)
collada-dom2.4-dp_2.4.4.0-ubuntu1~trusty1_amd64.deb (878 bytes)
--
◉ Kei Okada

youtalk, 2015-12-21:
After installing the packages, I have no problem.
Thank you very much for your help.

closed 2015-12-21 by youtalk

k-okada, 2015-12-21:
that's good to know