start-jsk/rtmros_hironx #423: 宿題

k-okada, 2016-01-22:
１。普通にlaunchを立ち上げるとcollisiondetectorのコンポーネントが入らないです。なので入れるようにするにはどうしたらいいかしらべて、入れてみてください。
２。その時に両腕の前腕がセルフコリジョンする動作をつくってください。いきなりだと危ないので、まず１０cmぐらい離れたし姿勢になって、１cmづつ、ちょっとづつ近づくようにしてみてください。
３。たぶん、手のモデルが入っていないので、手のモデルをいれてください。BoundingBoxでよいです。ちょっとおおきいぐらいでよいです。１、２cmぐらい。で、両腕の手と手がぶつかる動作を２と同じようにして動かして見てください。

できたら実機で試しましょう。
@pazeshun

k-okada, 2016-01-22:
http://www.airbusgroup.com/int/en/innovation-citizenship/Airbus-Shopfloor-Challenge-2016.html

pazeshun, 2016-01-22:
http://www.ros.org/news/2013/01/tecnalia-wins-the-factory-of-the-future-award-at-european-manufacturing-awards-2012-with-a-robot-pro.html

pazeshun, 2016-01-22:
http://www.airbus.com/presscentre/pressreleases/press-release-detail/detail/airbus-invests-in-robotics-as-part-of-its-commitment-to-innovation/

pazeshun, 2016-01-27:
[image]
一応、左手と右手が衝突するような目標座標を与えて動かしてやると、このようにモデルが接触した時点で止まるようになりました。

pazeshun, 2016-01-27:
rtmros_hironx/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.pyの、248行目
```
 # ['co', "CollisionDetector"],
```
のコメントアウトを外して
```
['co', "CollisionDetector"],
```
としてみて実行しただけです。

pazeshun, 2016-01-27:
とりあえず、実機につなげるかどうかだけ試したのですが、
```
$ ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host hiro014
```
を実行した所、変更前には出なかったエラーが２つ出ました。以下はログ全体です。
```
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with hiro014:15005
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7f00f7484dd0> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f00f7484dd0> isActive? = True 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7f00f7484a28>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7f00f7484d40>
[hrpsys.py]    version  =  1.0
;;
;; Loading ImpedanceController < 315.1.9
;;
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7f00f748f1b8> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f00f748f1b8> isActive? = True 
[hrpsys.py] simulation_mode : False
[hrpsys.py] creating components
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")
RTC named "seq" already exists.
[hrpsys.py] create Comp -> SequencePlayer : <hrpsys.rtm.RTcomponent instance at 0x7f00e1131bd8> (1.0)
[hrpsys.py] create CompSvc -> SequencePlayer Service : <OpenHRP._objref_SequencePlayerService instance at 0x7f00e1122ef0>
[hrpsys.py]   eval : [self.sh, self.sh_svc, self.sh_version] = self.createComp("StateHolder","sh")
RTC named "sh" already exists.
[hrpsys.py] create Comp -> StateHolder : <hrpsys.rtm.RTcomponent instance at 0x7f00e1131f80> (1.0)
[hrpsys.py] create CompSvc -> StateHolder Service : <OpenHRP._objref_StateHolderService instance at 0x7f00e11540e0>
[hrpsys.py]   eval : [self.fk, self.fk_svc, self.fk_version] = self.createComp("ForwardKinematics","fk")
RTC named "fk" already exists.
[hrpsys.py] create Comp -> ForwardKinematics : <hrpsys.rtm.RTcomponent instance at 0x7f00e11357a0> (1.0)
[hrpsys.py] create CompSvc -> ForwardKinematics Service : <OpenHRP._objref_ForwardKinematicsService instance at 0x7f00f7a0a758>
[hrpsys.py]   eval : [self.ic, self.ic_svc, self.ic_version] = self.createComp("ImpedanceController","ic")
RTC named "ic" already exists.
[hrpsys.py] create Comp -> ImpedanceController : <hrpsys.rtm.RTcomponent instance at 0x7f00e1131710> (1.0)
[hrpsys.py] create CompSvc -> ImpedanceController Service : <OpenHRP._objref_ImpedanceControllerService instance at 0x7f00e111ed88>
[hrpsys.py]   eval : [self.el, self.el_svc, self.el_version] = self.createComp("SoftErrorLimiter","el")
RTC named "el" already exists.
[hrpsys.py] create Comp -> SoftErrorLimiter : <hrpsys.rtm.RTcomponent instance at 0x7f00e1131050> (1.0)
[hrpsys.py] create CompSvc -> SoftErrorLimiter Service : <OpenHRP._objref_SoftErrorLimiterService instance at 0x7f00e1191050>
[hrpsys.py]   eval : [self.co, self.co_svc, self.co_version] = self.createComp("CollisionDetector","co")
[hrpsys.py] Fail to createComps CORBA.UNKNOWN(omniORB.UNKNOWN_UserException, CORBA.COMPLETED_MAYBE)
[hrpsys.py]   eval : [self.sc, self.sc_svc, self.sc_version] = self.createComp("ServoController","sc")
RTC named "sc" already exists.
[hrpsys.py] create Comp -> ServoController : <hrpsys.rtm.RTcomponent instance at 0x7f00e114e290> (1.0)
[hrpsys.py] create CompSvc -> ServoController Service : <OpenHRP._objref_ServoControllerService instance at 0x7f00e112c368>
[hrpsys.py]   eval : [self.log, self.log_svc, self.log_version] = self.createComp("DataLogger","log")
RTC named "log" already exists.
[hrpsys.py] create Comp -> DataLogger : <hrpsys.rtm.RTcomponent instance at 0x7f00f7a04560> (1.0)
[hrpsys.py] create CompSvc -> DataLogger Service : <OpenHRP._objref_DataLoggerService instance at 0x7f00e11283f8>
[hrpsys.py]   eval : [self.rmfo, self.rmfo_svc, self.rmfo_version] = self.createComp("AbsoluteForceSensor","rmfo")
RTC named "rmfo" already exists.
[hrpsys.py] create Comp -> AbsoluteForceSensor : <hrpsys.rtm.RTcomponent instance at 0x7f00e1119758> (1.0)
[hrpsys.py] create CompSvc -> AbsoluteForceSensor Service : <OpenHRP._objref_AbsoluteForceSensorService instance at 0x7f00e1135a70>
[hrpsys.py] connecting components
[rtm.py]    Connect sh.qOut - ic.qRef
[... 145 lines cut ...]
[hrpsys.py] sensor names for DataLogger
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = qOut to sh_qOut
[rtm.py]    Connect sh.qOut - log.sh_qOut
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = tqOut to sh_tqOut
[rtm.py]    Connect sh.tqOut - log.sh_tqOut
[hrpsys.py]   setupLogger : record type = TimedPoint3D, name = basePosOut to sh_basePosOut
[rtm.py]    Connect sh.basePosOut - log.sh_basePosOut
[hrpsys.py]   setupLogger : record type = TimedOrientation3D, name = baseRpyOut to sh_baseRpyOut
[rtm.py]    Connect sh.baseRpyOut - log.sh_baseRpyOut
[hrpsys.py]   setupLogger : record type = TimedPoint3D, name = zmpOut to sh_zmpOut
[rtm.py]    Connect sh.zmpOut - log.sh_zmpOut
[... cut ...]
[rtm.py]    Connect RobotHardware0.emergencySignal - log.emergencySignal
[hrpsys.py]   setupLogger : record type = TimedLongSeqSeq, name = servoState to RobotHardware0_servoState
[rtm.py]    Failed to connect RobotHardware0.servoState to None([None])
[hrpsys.py] setup joint groups
[hrpsys.py] initialized successfully
[rtm.py]    Connect RobotHardware0.lhsensor - ic.lhsensor
[rtm.py]    Connect RobotHardware0.rhsensor - ic.rhsensor
[ros_client] ros master is not running, so do not create ros client...
```

pazeshun, 2016-01-27:
以下は、変更前に戻した時のログです。
```
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with hiro014:15005
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7fc7e3d90dd0> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7fc7e3d90dd0> isActive? = True 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7fc7e3d90a28>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7fc7e3d90d40>
[hrpsys.py]    version  =  1.0
;;
;; Loading ImpedanceController < 315.1.9
;;
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7fc7e3d9b1b8> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7fc7e3d9b1b8> isActive? = True 
[hrpsys.py] simulation_mode : False
[hrpsys.py] creating components
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")
RTC named "seq" already exists.
[hrpsys.py] create Comp -> SequencePlayer : <hrpsys.rtm.RTcomponent instance at 0x7fc7cda3dbd8> (1.0)
[hrpsys.py] create CompSvc -> SequencePlayer Service : <OpenHRP._objref_SequencePlayerService instance at 0x7fc7cda2eef0>
[hrpsys.py]   eval : [self.sh, self.sh_svc, self.sh_version] = self.createComp("StateHolder","sh")
RTC named "sh" already exists.
[hrpsys.py] create Comp -> StateHolder : <hrpsys.rtm.RTcomponent instance at 0x7fc7cda3df80> (1.0)
[hrpsys.py] create CompSvc -> StateHolder Service : <OpenHRP._objref_StateHolderService instance at 0x7fc7cda60050>
[hrpsys.py]   eval : [self.fk, self.fk_svc, self.fk_version] = self.createComp("ForwardKinematics","fk")
RTC named "fk" already exists.
[hrpsys.py] create Comp -> ForwardKinematics : <hrpsys.rtm.RTcomponent instance at 0x7fc7cda41b90> (1.0)
[hrpsys.py] create CompSvc -> ForwardKinematics Service : <OpenHRP._objref_ForwardKinematicsService instance at 0x7fc7e4316560>
[hrpsys.py]   eval : [self.ic, self.ic_svc, self.ic_version] = self.createComp("ImpedanceController","ic")
RTC named "ic" already exists.
[hrpsys.py] create Comp -> ImpedanceController : <hrpsys.rtm.RTcomponent instance at 0x7fc7cda3d1b8> (1.0)
[hrpsys.py] create CompSvc -> ImpedanceController Service : <OpenHRP._objref_ImpedanceControllerService instance at 0x7fc7cda2a3b0>
[hrpsys.py]   eval : [self.el, self.el_svc, self.el_version] = self.createComp("SoftErrorLimiter","el")
RTC named "el" already exists.
[hrpsys.py] create Comp -> SoftErrorLimiter : <hrpsys.rtm.RTcomponent instance at 0x7fc7cda3d2d8> (1.0)
[hrpsys.py] create CompSvc -> SoftErrorLimiter Service : <OpenHRP._objref_SoftErrorLimiterService instance at 0x7fc7cda9d050>
[hrpsys.py]   eval : [self.sc, self.sc_svc, self.sc_version] = self.createComp("ServoController","sc")
RTC named "sc" already exists.
[hrpsys.py] create Comp -> ServoController : <hrpsys.rtm.RTcomponent instance at 0x7fc7cda5a518> (1.0)
[hrpsys.py] create CompSvc -> ServoController Service : <OpenHRP._objref_ServoControllerService instance at 0x7fc7e3d90b90>
[hrpsys.py]   eval : [self.log, self.log_svc, self.log_version] = self.createComp("DataLogger","log")
RTC named "log" already exists.
[hrpsys.py] create Comp -> DataLogger : <hrpsys.rtm.RTcomponent instance at 0x7fc7cda41950> (1.0)
[hrpsys.py] create CompSvc -> DataLogger Service : <OpenHRP._objref_DataLoggerService instance at 0x7fc7cda34830>
[hrpsys.py]   eval : [self.rmfo, self.rmfo_svc, self.rmfo_version] = self.createComp("AbsoluteForceSensor","rmfo")
RTC named "rmfo" already exists.
[hrpsys.py] create Comp -> AbsoluteForceSensor : <hrpsys.rtm.RTcomponent instance at 0x7fc7cdaa57a0> (1.0)
[hrpsys.py] create CompSvc -> AbsoluteForceSensor Service : <OpenHRP._objref_AbsoluteForceSensorService instance at 0x7fc7e4310128>
[hrpsys.py] connecting components
[rtm.py]    Connect sh.qOut - ic.qRef
[rtm.py]    Connect ic.q - el.qRef
[rtm.py]    Connect el.q - RobotHardware0.qRef
[rtm.py]    Connect RobotHardware0.servoState - el.servoStateIn
[rtm.py]    Connect RobotHardware0.q - sh.currentQIn
[rtm.py]    Connect RobotHardware0.q - fk.q
[rtm.py]    Connect sh.qOut - fk.qRef
[... 27 lines cut ...]
[hrpsys.py]   setupLogger : tau arleady exists in DataLogger
[rtm.py]    Connect RobotHardware0.tau - log.RobotHardware0_tau
[hrpsys.py] sensor names for DataLogger
[hrpsys.py]   setupLogger : qOut arleady exists in DataLogger
[rtm.py]    Connect sh.qOut - log.sh_qOut
[hrpsys.py]   setupLogger : tqOut arleady exists in DataLogger
[rtm.py]    Connect sh.tqOut - log.sh_tqOut
[hrpsys.py]   setupLogger : basePosOut arleady exists in DataLogger
[rtm.py]    Connect sh.basePosOut - log.sh_basePosOut
[hrpsys.py]   setupLogger : baseRpyOut arleady exists in DataLogger
[rtm.py]    Connect sh.baseRpyOut - log.sh_baseRpyOut
[hrpsys.py]   setupLogger : zmpOut arleady exists in DataLogger
[rtm.py]    Connect sh.zmpOut - log.sh_zmpOut
[hrpsys.py]   setupLogger : q arleady exists in DataLogger
[... cut ...]
[rtm.py]    Connect el.q - log.el_q
[hrpsys.py]   setupLogger : emergencySignal arleady exists in DataLogger
[rtm.py]    Connect RobotHardware0.emergencySignal - log.emergencySignal
[hrpsys.py]   setupLogger : record type = TimedLongSeqSeq, name = servoState to RobotHardware0_servoState
[rtm.py]    Failed to connect RobotHardware0.servoState to None([None])
[hrpsys.py] setup joint groups
[hrpsys.py] initialized successfully
[ros_client] ros master is not running, so do not create ros client...
```

k-okada, 2016-01-29:
qnx側の/opt/jsk
かな？のしたにCollisionDetector.soがあるかどうか見てみてくダサい．例えば，DataLogger.soがあるディレクトリと同じ所に有るはずです．

pazeshun, 2016-01-29:
```
$ find /opt/jsk -name DataLogger.so
/opt/jsk/src/hrpsys-base-source-315.1.7/rtc/DataLogger/CMakeFiles/CMakeRelink.dir/DataLogger.so
/opt/jsk/src/hrpsys-base-source-315.1.7/lib/DataLogger.so
/opt/jsk/lib/DataLogger.so
```
となり、どのディレクトリを調べてもCollisionDetector.soらしきものはありませんでした

pazeshun, 2016-01-29:
`/opt/jsk/src/hrpsys-base-source-315.1.7/rtc/CollisionDetector`はありましたが、その中にCMakeFilesはありませんでした。
これ、コンパイルするとできるってことなんですかね

k-okada, 2016-01-29:
/opt/jsk のどっかにbuildというディレクトリかCMakefilesが有るディレクトリがあるんだけどわかるかな．

pazeshun, 2016-02-09:
```
$ pwd                             
/opt/jsk
$ find . -type d -name CMakeFiles 
./src/hrpsys-base-source-315.1.7/test/CMakeFiles
./src/hrpsys-base-source-315.1.7/python/CMakeFiles
./src/hrpsys-base-source-315.1.7/sample/HRP4C/data/CMakeFiles
./src/hrpsys-base-source-315.1.7/sample/HRP4C/CMakeFiles
./src/hrpsys-base-source-315.1.7/sample/PA10/CMakeFiles
./src/hrpsys-base-source-315.1.7/sample/CMakeFiles
./src/hrpsys-base-source-315.1.7/ec/hrpEC/CMakeFiles
./src/hrpsys-base-source-315.1.7/ec/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/Range2PointCloud/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/GraspController/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/TorqueController/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/ThermoLimiter/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/ThermoEstimator/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/AbsoluteForceSensor/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/NullComponent/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/ServoController/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/HGcontroller/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/RobotHardware/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/KalmanFilter/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/DataLogger/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/SoftErrorLimiter/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/TorqueFilter/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/SequencePlayer/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/ImpedanceController/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/VirtualForceSensor/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/AutoBalancer/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/Stabilizer/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/WavPlayer/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/ForwardKinematics/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/StateHolder/CMakeFiles
./src/hrpsys-base-source-315.1.7/rtc/CMakeFiles
./src/hrpsys-base-source-315.1.7/lib/io/CMakeFiles
./src/hrpsys-base-source-315.1.7/lib/CMakeFiles
./src/hrpsys-base-source-315.1.7/idl/CMakeFiles
./src/hrpsys-base-source-315.1.7/CMakeFiles
./src/hrpsys-base-source-315.1.7/CMakeFiles/CMakeTmp/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/hrpCorba/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/hrpCollision/Opcode/Ice/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/hrpCollision/Opcode/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/hrpCollision/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/hrpUtil/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/hrpModel/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/hrpRTM/CMakeFiles
./src/OpenHRP-3.1.5/hrplib/CMakeFiles
./src/OpenHRP-3.1.5/sample/JoystickControl/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SampleHG/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/PD_HGtest/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SampleCrawler/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SampleLF/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SamplePD_HG/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/PA10Controller/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SampleController/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SamplePD/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SampleSV/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/SampleRH2/CMakeFiles
./src/OpenHRP-3.1.5/sample/controller/CMakeFiles
./src/OpenHRP-3.1.5/sample/example/clap/CMakeFiles
./src/OpenHRP-3.1.5/sample/example/move_ankle/CMakeFiles
./src/OpenHRP-3.1.5/sample/example/customizer/CMakeFiles
./src/OpenHRP-3.1.5/sample/example/viewSimulator/CMakeFiles
./src/OpenHRP-3.1.5/sample/example/simulationEC/CMakeFiles
./src/OpenHRP-3.1.5/sample/example/scheduler/CMakeFiles
./src/OpenHRP-3.1.5/sample/example/CMakeFiles
./src/OpenHRP-3.1.5/sample/CMakeFiles
./src/OpenHRP-3.1.5/script/CMakeFiles
./src/OpenHRP-3.1.5/server/CollisionDetector/CMakeFiles
./src/OpenHRP-3.1.5/server/ModelLoader/CMakeFiles
./src/OpenHRP-3.1.5/server/AistDynamicsSimulator/CMakeFiles
./src/OpenHRP-3.1.5/server/CMakeFiles
./src/OpenHRP-3.1.5/idl/OpenRTM/1.1.0/CMakeFiles
./src/OpenHRP-3.1.5/idl/OpenRTM/CMakeFiles
./src/OpenHRP-3.1.5/idl/OpenHRP/CMakeFiles
./src/OpenHRP-3.1.5/idl/CMakeFiles
./src/OpenHRP-3.1.5/CMakeFiles
./src/OpenHRP-3.1.5/CMakeFiles/CMakeTmp/CMakeFiles
```
という出力が得られました。意味が違ってたらごめんなさい。

pazeshun, 2016-02-19:
こないだ、
```
# make install
```
がうまく行ったと思っていたのですが、再起動を忘れていたためのようで、起動してPCから接続しようとすると以下のようなログが出てきて接続が失敗しました。
```
$ ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host hiro014
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with hiro014:15005
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : None ( timeout 0 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 1 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 2 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 3 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 4 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 5 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 6 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 7 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 8 < 10)
[hrpsys.py] wait for RobotHardware0 : None ( timeout 9 < 10)
[hrpsys.py] Could not find RobotHardware0
---------------------------------------------------------------------------
TRANSIENT                                 Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, *where)
    202             else:
    203                 filename = fname
--> 204             __builtin__.execfile(filename, *where)

/opt/ros/indigo/share/hironx_ros_bridge/scripts/hironx.py in <module>()
     73         args.modelfile = unknown[1]
     74     robot = hiro = hironx_client.HIRONX()
---> 75     robot.init(robotname=args.robot, url=args.modelfile)
     76 
     77     # ROS Client

/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/hironx_client.pyc in init(self, robotname, url)
    157         # reload for hrpsys 315.1.8
    158         HrpsysConfigurator.waitForModelLoader(self)
--> 159         HrpsysConfigurator.waitForRTCManagerAndRoboHardware(self, robotname=robotname)
    160         print self.configurator_name, "Hrpsys controller version info: "
    161         if self.ms :

/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py in waitForRTCManagerAndRoboHardware(self, robotname, managerhost)
    914         '''
    915         self.waitForRTCManager(managerhost)
--> 916         self.waitForRobotHardware(robotname)
    917         self.checkSimulationMode()
    918 

/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py in waitForRobotHardware(self, robotname)
    881             print(self.configurator_name + "Could not find " + robotname)
    882             if self.ms:
--> 883                 print(self.configurator_name + "Candidates are .... " + str([x.name()  for x in self.ms.get_components()]))
    884             print(self.configurator_name + "Exitting.... " + robotname)
    885             exit(1)

/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/rtm.py in get_components(self)
    251     def get_components(self):
    252         cs = []
--> 253         crefs = self.ref.get_components()
    254         for cref in crefs:
    255             c = RTcomponent(cref)

/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/Manager_idl.pyc in get_components(self, *args)
    163 
    164     def get_components(self, *args):
--> 165         return _omnipy.invoke(self, "get_components", _0_RTM.Manager._d_get_components, args)
    166 
    167     def get_component_profiles(self, *args):

TRANSIENT: CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
```
この時、Hiroにsshでログインしてログを確認すると、以下のようになってました。
```
$ cat jsk/var/log/rtcd.log 
Logger::Logger: streambuf address = 0x805fc70
Memory fault (core dumped) 
$ pwd                       
/opt
```
なお、`# make install`前にコピーしておいたjskディレクトリと、現在のjskディレクトリを入れ替えて再起動すると、Hiro内部のログは以下のようになっていました。
```
$ cat /opt/jsk/var/log/rtcd.log 
Logger::Logger: streambuf address = 0x805fc70
hrpExecutionContext is registered
pdgains.file_name: /opt/jsk/etc/HIRONX/hrprtc/PDgains.sav
dof = 15
client: msg_no: _IO_MAX + 4
client: msg_reply:
rom # 4, soft # 300, 28/9/2011
serial # 5171, model # 129, cal day 28/9/2011
Newton, Newton*meter*10, mm*10, 14 bits, 255 ch
(Crom # 4, soft # 300, 4/10/2011
serial # 5172, model # 129, cal day 4/10/2011
Newton, Newton*meter*10, mm*10, 14 bits, 255 ch
(C
open_iob - shmif instance at 0x80ca0e8
the number of gyros = 0
the number of accelerometers = 0
the number of force sensors = 2
lhsensor
rhsensor
period = 5[ms], priority = 49
```
この時は、おそらく問題なく起動できているものと思われます。

k-okada, 2016-02-19:
お，古いディレクトリに新しい所に出来たcollision detctorだけコピーするとどうなるかな．

k-okada, 2016-02-19:
後，古いディレクトリと新しいディレクトリで diff -r したい

pazeshun, 2016-02-19:
```
$ ldd /opt/jsk.20160219/lib/libhrpIo.so 
/opt/jsk.20160219/lib/libhrpIo.so:
	libhrpIo.so => /opt/jsk.20160219/lib/libhrpIo.so (0xb8200000)
	libboost_filesystem.so => /usr/pkg/lib/libboost_filesystem.so.1.42.0 (0xb820d000)
	libboost_thread.so => /usr/pkg/lib/libboost_thread.so.1.42.0 (0xb8225000)
	libboost_regex.so => /usr/pkg/lib/libboost_regex.so.1.42.0 (0xb823a000)
	libf2c.so => /usr/pkg/lib/libf2c.so (0xb8320000)
	libcpp.so.4 => /lib/libcpp.so.4 (0xb833d000)
	libboost_system.so => /usr/pkg/lib/libboost_system.so.1.42.0 (0xb83a3000)
	libm.so.2 => /lib/libm.so.2 (0xb83a7000)
$ ldd /opt/jsk/lib/libhrpIo.so           
/opt/jsk/lib/libhrpIo.so:
	libhrpIo.so => /opt/jsk/lib/libhrpIo.so (0xb8200000)
	libboost_system.so => /usr/pkg/lib/libboost_system.so.1.42.0 (0xb8219000)
	libboost_signals.so => /usr/pkg/lib/libboost_signals.so.1.42.0 (0xb821d000)
	libboost_filesystem.so => /usr/pkg/lib/libboost_filesystem.so.1.42.0 (0xb823a000)
	libboost_thread.so => /usr/pkg/lib/libboost_thread.so.1.42.0 (0xb8252000)
	libboost_regex.so => /usr/pkg/lib/libboost_regex.so.1.42.0 (0xb8267000)
	libf2c.so => /usr/pkg/lib/libf2c.so.0 (0xb834d000)
	libcpp.so.4 => /lib/libcpp.so.4 (0xb836a000)
	libm.so.2 => /lib/libm.so.2 (0xb83d0000)
```

pazeshun, 2016-02-19:
/opt/jsk/src/hrpsys-base-source-315.1.7/cmake_modules/FindQuickHull.cmakeを以下のように編集してみました。
```
FIND_PATH(
QHULL_DIR
# NAMES include/qhull/qhull.h
# PATHS /usr /usr/local
NAMES include/libqhull/libqhull.h
PATHS /usr /usr/local /opt/jsk
DOC "the top directory of qhull")

IF ( QHULL_DIR )
    MESSAGE(STATUS "Found Qhull in ${QHULL_DIR}")
    set(QHULL_INCLUDE_DIR ${QHULL_DIR}/include)
    if (APPLE)
      set(QHULL_LIBRARIES qhull)
    else()
      set(QHULL_LIBRARIES qhull)
    endif()
    set(QHULL_FOUND true)
ELSE ( QHULL_INCLUDE_DIR )
    MESSAGE(STATUS "Qhull not found")
    set(QHULL_FOUND false)
ENDIF ( QHULL_DIR )
```
で、cmakeしようとしたのですが、以下のようなエラーが出てきました。
```
$ cmake /opt/jsk/src/hrpsys-base-source-315.1.7
-- Found OpenRTM-aist 1.1.0 in /opt/jsk
Package openhrp3.1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `openhrp3.1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openhrp3.1' found
Package openhrp3.1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `openhrp3.1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openhrp3.1' found
Package openhrp3.1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `openhrp3.1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openhrp3.1' found
Package openhrp3.1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `openhrp3.1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openhrp3.1' found
CMake Error at cmake_modules/FindOpenHRP.cmake:76 (message):
  OpenHRP required, please specify it's location.
Call Stack (most recent call first):
  CMakeLists.txt:38 (find_package)

-- Configuring incomplete, errors occurred!
```

k-okada, 2016-02-19:
echo $PKG_CONFIG_PATH
して，optがない気がするので，

export PKG_CONFIG_PATH=/opt/jsk/lib/pkgconfig:/usr/pkg/lib/pkgconfig;

みたいなかんじかな．
--
◉ Kei Okada

pazeshun, 2016-02-19:
`echo $PKG_CONFIG_PATH`しても何も出てこなかったので、
```
$ export PKG_CONFIG_PATH=/opt/jsk/lib/pkgconfig:/usr/pkg/lib/pkgconfig; 
$ cmake /opt/jsk/src/hrpsys-base-source-315.1.7
```
すると、
```
$ make
[ 32%] Built target hrpsysBaseStub
[ 32%] Built target hrpIo
[ 33%] Built target hrpEC
[ 34%] Built target NullComponent
[ 36%] Built target NullComponentComp
[ 37%] Built target RobotHardware
[ 39%] Built target RobotHardwareComp
[ 40%] Built target StateHolder
[ 42%] Built target StateHolderComp
[ 43%] Built target WavPlayer
[ 44%] Built target WavPlayerComp
[ 46%] Built target SequencePlayer
[ 49%] Built target SequencePlayerComp
[ 50%] Built target DataLogger
[ 51%] Built target DataLoggerComp
[ 52%] Built target ForwardKinematics
[ 53%] Built target ForwardKinematicsComp
[ 54%] Built target HGcontroller
[ 55%] Built target HGcontrollerComp
[ 55%] Built target Range2PointCloud
[ 56%] Built target Range2PointCloudComp
[ 57%] Built target ImpedanceController
[ 59%] Built target ImpedanceControllerComp
[ 62%] Built target AutoBalancer
[ 65%] Built target AutoBalancerComp
[ 67%] Built target testGaitGenerator
[ 68%] Built target testPreviewController
[ 70%] Built target SoftErrorLimiter
[ 72%] Built target SoftErrorLimiterComp
[ 73%] Built target VirtualForceSensor
[ 74%] Built target VirtualForceSensorComp
[ 75%] Built target GraspController
[ 76%] Built target GraspControllerComp
[ 77%] Built target TorqueFilter
[ 78%] Built target TorqueFilterComp
[ 79%] Built target KalmanFilter
[ 80%] Built target KalmanFilterComp
[ 82%] Built target Stabilizer
[ 85%] Built target StabilizerComp
[ 86%] Built target AbsoluteForceSensor
[ 87%] Built target AbsoluteForceSensorComp
[ 88%] Built target ServoController
[ 89%] Built target ServoControllerComp
[ 90%] Built target testServoSerial
[ 90%] Built target ThermoEstimator
[ 91%] Built target ThermoEstimatorComp
[ 92%] Built target ThermoLimiter
[ 94%] Built target ThermoLimiterComp
[ 96%] Built target TorqueController
[ 98%] Built target TorqueControllerComp
[100%] Built target testMotorTorqueController
```
となります。これ、CollisionDetector入ってないですよね

k-okada, 2016-02-19:
cwd はどこになっている．既に一回cmakeされているかな？
catkin b --force-cmake 的なことをやらなければいけないので，
CMakeFiles みたいなデレクトリの中のCollisionDetector のディレクトリを消してみて下さい．わからんければ
cmake を打ったディレクトリで find | grep Collision してみて

pazeshun, 2016-02-19:
```
$ pwd
/opt/jsk/src/hrpsys-base-source-315.1.7
$ find | grep Collision  
./rtc/CollisionDetector
./rtc/CollisionDetector/.svn
./rtc/CollisionDetector/.svn/text-base
./rtc/CollisionDetector/.svn/text-base/CollisionDetectorComp.cpp.svn-base
./rtc/CollisionDetector/.svn/text-base/CollisionDetector.cpp.svn-base
./rtc/CollisionDetector/.svn/text-base/GLscene.h.svn-base
./rtc/CollisionDetector/.svn/text-base/VclipLinkPair.cpp.svn-base
./rtc/CollisionDetector/.svn/text-base/CollisionDetectorViewer.cpp.svn-base
./rtc/CollisionDetector/.svn/text-base/CollisionDetectorService_impl.cpp.svn-base
./rtc/CollisionDetector/.svn/text-base/SetupCollisionPair.cpp.svn-base
./rtc/CollisionDetector/.svn/text-base/TimedPosture.h.svn-base
./rtc/CollisionDetector/.svn/text-base/CollisionDetector.h.svn-base
./rtc/CollisionDetector/.svn/text-base/CMakeLists.txt.svn-base
./rtc/CollisionDetector/.svn/text-base/CollisionDetectorService_impl.h.svn-base
./rtc/CollisionDetector/.svn/text-base/VclipLinkPair.h.svn-base
./rtc/CollisionDetector/.svn/text-base/GLscene.cpp.svn-base
./rtc/CollisionDetector/.svn/prop-base
./rtc/CollisionDetector/.svn/props
./rtc/CollisionDetector/.svn/tmp
./rtc/CollisionDetector/.svn/tmp/text-base
./rtc/CollisionDetector/.svn/tmp/prop-base
./rtc/CollisionDetector/.svn/tmp/props
./rtc/CollisionDetector/.svn/dir-prop-base
./rtc/CollisionDetector/.svn/all-wcprops
./rtc/CollisionDetector/.svn/entries
./rtc/CollisionDetector/vclip_1.0
./rtc/CollisionDetector/vclip_1.0/.svn
./rtc/CollisionDetector/vclip_1.0/.svn/text-base
./rtc/CollisionDetector/vclip_1.0/.svn/text-base/Readme.svn-base
./rtc/CollisionDetector/vclip_1.0/.svn/text-base/Makefile.svn-base
./rtc/CollisionDetector/vclip_1.0/.svn/text-base/ii_files.svn-base
./rtc/CollisionDetector/vclip_1.0/.svn/prop-base
./rtc/CollisionDetector/vclip_1.0/.svn/props
./rtc/CollisionDetector/vclip_1.0/.svn/tmp
./rtc/CollisionDetector/vclip_1.0/.svn/tmp/text-base
./rtc/CollisionDetector/vclip_1.0/.svn/tmp/prop-base
./rtc/CollisionDetector/vclip_1.0/.svn/tmp/props
./rtc/CollisionDetector/vclip_1.0/.svn/all-wcprops
./rtc/CollisionDetector/vclip_1.0/.svn/entries
./rtc/CollisionDetector/vclip_1.0/ptrees
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/ring.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/sixspike.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/ring2.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/mitsu.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/grid.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/sphere.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/jack.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/simple.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/Readme.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/platonics.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/text-base/platonics2.pt.svn-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/prop-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/props
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/tmp
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/tmp/text-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/tmp/prop-base
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/tmp/props
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/all-wcprops
./rtc/CollisionDetector/vclip_1.0/ptrees/.svn/entries
./rtc/CollisionDetector/vclip_1.0/ptrees/ring.pt
./rtc/CollisionDetector/vclip_1.0/ptrees/sixspike.pt
./rtc/CollisionDetector/vclip_1.0/ptrees/ring2.pt
./rtc/CollisionDetector/vclip_1.0/ptrees/mitsu.pt
./rtc/CollisionDetector/vclip_1.0/ptrees/grid.pt
[... 160 lines cut ...]
./rtc/CollisionDetector/vclip_1.0/examples/ap1.txt
./rtc/CollisionDetector/vclip_1.0/Readme
./rtc/CollisionDetector/vclip_1.0/Makefile
./rtc/CollisionDetector/vclip_1.0/ii_files
./rtc/CollisionDetector/CollisionDetectorComp.cpp
./rtc/CollisionDetector/CollisionDetector.cpp
./rtc/CollisionDetector/GLscene.h
./rtc/CollisionDetector/VclipLinkPair.cpp
./rtc/CollisionDetector/CollisionDetectorViewer.cpp
./rtc/CollisionDetector/CollisionDetectorService_impl.cpp
./rtc/CollisionDetector/SetupCollisionPair.cpp
./rtc/CollisionDetector/TimedPosture.h
./rtc/CollisionDetector/CollisionDetector.h
./rtc/CollisionDetector/CMakeLists.txt
./rtc/CollisionDetector/CollisionDetectorService_impl.h
./rtc/CollisionDetector/VclipLinkPair.h
./rtc/CollisionDetector/GLscene.cpp
./idl/.svn/text-base/CollisionDetectorService.idl.svn-base
./idl/CollisionDetectorService.idl
./idl/CMakeFiles/hrpsysBaseStub.dir/CollisionDetectorServiceSk.o
./idl/CMakeFiles/hrpsysBaseStub.dir/CollisionDetectorServiceDynSk.o
./idl/CollisionDetectorService.hh
./idl/CollisionDetectorServiceSk.cpp
./idl/CollisionDetectorServiceDynSk.cpp
```

k-okada, 2016-02-19:
cmakeってコマンドを打ったディレクトリでお願いします

pazeshun, 2016-02-19:
そこで打ちました

k-okada, 2016-02-19:
CMakeFilesみたいなディレクトリない？

pazeshun, 2016-02-19:
```
$ pwd
/opt/jsk/src/hrpsys-base-source-315.1.7
$ cd CMakeFiles/
$ ls -l
total 394
-rw-r--r--  1 hiro      users          1259 Jul 09  2014 CMakeCCompiler.cmake
-rw-r--r--  1 hiro      users          1375 Jul 09  2014 CMakeCXXCompiler.cmake
-rwxrwxr-x  1 hiro      users          5045 Jul 09  2014 CMakeDetermineCompilerABI_C.bin
-rwxrwxr-x  1 hiro      users          5339 Jul 09  2014 CMakeDetermineCompilerABI_CXX.bin
-rw-rw-r--  1 hiro      users          1056 Jul 09  2014 CMakeDirectoryInformation.cmake
-rw-rw-rw-  1 hiro      users          3073 Jul 09  2014 CMakeError.log
-rw-rw-rw-  1 hiro      users         15908 Jul 09  2014 CMakeOutput.log
-rw-rw-r--  1 hiro      users          1526 Feb 19 19:38 CMakeRuleHashes.txt
-rw-r--r--  1 hiro      users           340 Jul 09  2014 CMakeSystem.cmake
drwxrwxr-x  3 hiro      users          1024 Jan 19  2014 CMakeTmp
drwxrwxr-x  2 hiro      users          1024 Jan 19  2014 CompilerIdC
drwxrwxr-x  2 hiro      users          1024 Jan 19  2014 CompilerIdCXX
-rw-rw-r--  1 hiro      users          9423 Feb 19 19:38 Makefile.cmake
-rw-rw-r--  1 hiro      users        145475 Feb 19 19:38 Makefile2
-rw-rw-r--  1 hiro      users          4589 Feb 19 19:38 TargetDirectories.txt
-rw-rw-r--  1 hiro      users            85 Feb 19 19:38 cmake.check_cache
-rw-rw-r--  1 hiro      users             4 Feb 19 19:38 progress.marks
```

k-okada, 2016-02-19:
$ cmake /opt/jsk/src/hrpsys-base-source-315.1.7
すたらなんてひょうじがでたの？

pazeshun, 2016-02-19:
```
$ cmake /opt/jsk/src/hrpsys-base-source-315.1.7
-- Found OpenRTM-aist 1.1.0 in /opt/jsk
-- Found OpenHRP 3.1.5 in /opt/jsk
-- Found Qhull in /opt/jsk
-- checking for module 'opencv'
--   package 'opencv' not found
-- checking for module 'glew'
--   package 'glew' not found
-- checking for module 'opencv-2.3.1'
--   package 'opencv-2.3.1' not found
CMake Warning at CMakeLists.txt:111 (message):
  opencv not found

Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/DataLoggerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/StateHolderService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/RobotHardwareService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/WavPlayerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/SequencePlayerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/ForwardKinematicsService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/ExecutionProfileService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/OGMap3DService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/NullService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/TimeKeeperService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/Img.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/HRPDataTypes.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/pointcloud.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/ImpedanceControllerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/SoftErrorLimiterService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/CollisionDetectorService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/GraspControllerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/TorqueFilterService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/KalmanFilterService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/VirtualForceSensorService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/StabilizerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/AutoBalancerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/AbsoluteForceSensorService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/ServoControllerService.idl exists
Check if /opt/jsk/src/hrpsys-base-source-315.1.7/idl/TorqueControllerService.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/ClockGenerator.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/CollisionDetector.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/Controller.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/DynamicsSimulator.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/ModelLoader.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/OnlineViewer.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/OpenHRPCommon.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/ViewSimulator.idl exists
Check if /opt/jsk/share/OpenHRP-3.1/idl/OpenHRP/World.idl exists
CMake Warning (dev) at ec/hrpEC/CMakeLists.txt:4 (link_directories):
  This command specifies the relative path

    io

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at rtc/RobotHardware/CMakeLists.txt:3 (link_directories):
  This command specifies the relative path

    io

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Irrlicht not found
-- checking for module 'octomap'
--   package 'octomap' not found
CMake Warning at test/CMakeLists.txt:29 (message):
  GTest not found; C++ tests will fail to build.

-- Configuring done
-- Generating done
-- Build files have been written to: /opt/jsk/src/hrpsys-base-source-315.1.7
```

k-okada, 2016-02-19:
CollisionDetectorのCMakeLists.txtに
message("hoge hoge")
と書いてデバッグかな。

pazeshun, 2016-02-19:
わかりました。
もう帰らないといけないので、来週やります。

k-okada, 2016-02-19:
はい。了解。

pazeshun, 2016-03-08:
```
$ cat /opt/jsk/src/hrpsys-base-source-315.1.7/rtc/CMakeLists.txt  
```
すると、
```
add_subdirectory(NullComponent)
add_subdirectory(RobotHardware)
add_subdirectory(StateHolder)
add_subdirectory(WavPlayer)
add_subdirectory(SequencePlayer)
add_subdirectory(DataLogger)
if (NOT QNXNTO)
  add_subdirectory(Joystick)
  add_subdirectory(Joystick2Velocity2D)
  add_subdirectory(Joystick2Velocity3D)
  add_subdirectory(Joystick2PanTiltAngles)
endif()
add_subdirectory(ForwardKinematics)
add_subdirectory(HGcontroller)
add_subdirectory(Range2PointCloud)
add_subdirectory(ImpedanceController)
add_subdirectory(AutoBalancer)
add_subdirectory(SoftErrorLimiter)
add_subdirectory(VirtualForceSensor)
add_subdirectory(GraspController)
add_subdirectory(TorqueFilter)
add_subdirectory(KalmanFilter)
add_subdirectory(Stabilizer)
add_subdirectory(AbsoluteForceSensor)
add_subdirectory(ServoController)
add_subdirectory(ThermoEstimator)
add_subdirectory(ThermoLimiter)
add_subdirectory(TorqueController)
if (USE_HRPSYSUTIL)
  add_subdirectory(Viewer)
  add_subdirectory(CameraImageViewer)
  add_subdirectory(JpegDecoder)
  add_subdirectory(VirtualCamera)
  add_subdirectory(Simulator)
  if (QHULL_FOUND)
    add_subdirectory(CollisionDetector)
  endif()
endif()

if (NOT APPLE AND USE_HRPSYSUTIL)
  add_subdirectory(VideoCapture)
endif()

find_package(Irrlicht)
if (IRRLICHT_FOUND AND USE_HRPSYSUTIL)
  add_subdirectory(OGMap3DViewer)
endif()

# Octomap
pkg_check_modules(OCTOMAP octomap)
if (OCTOMAP_FOUND)
  add_subdirectory(OccupancyGridMap3D)
endif()

set(EXTRA_RTC_DIRS "" CACHE PATH "directories of extra RTCs")
foreach(rtc_dir ${EXTRA_RTC_DIRS})
  add_subdirectory(${rtc_dir} ${rtc_dir})
endforeach()
```
と出てきました。これをみると、USE_HRPSYSUTILがtrueでないと、CollisionDetectorが読み込まれないことがわかります。
```
$ cat /opt/jsk/src/hrpsys-base-source-315.1.7/CMakeLists.txt
```
すると、
```
[... 135 lines cut ...]
  DESTINATION lib/pkgconfig)

if(COMPILE_JAVA_STUFF)
  add_subdirectory(jython)
endif()
add_subdirectory(python)
add_subdirectory(idl)
add_subdirectory(lib)
add_subdirectory(ec)
add_subdirectory(rtc)
option(ENABLE_DOXYGEN "Use Doxygen" ON)
if(ENABLE_DOXYGEN)
  add_subdirectory(doc)
endif()
if (NOT QNXNTO)
  add_subdirectory(util)
endif()
add_subdirectory(sample)
add_subdirectory(test)
```
となります、USE_HRPSYSUTILがtrueになる条件は、OpenCVやOpenGL、GLEW、GLUT、SDLが存在することであるとわかります。
```
$ cmake /opt/jsk/src/hrpsys-base-source-315.1.7
```
した時の表示を読むと、OpenCVとかが見つかっていないことがわかるので、これが原因でCollision Detectorがビルドできないことがわかりました。

pazeshun, 2016-09-06:
現在、`make`を行った時に出てくるエラーは以下のようになります。
```
Linking CXX shared library ../../lib/CollisionDetector.so
[ 96%] Built target CollisionDetector
Scanning dependencies of target CollisionDetectorComp
[ 97%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/CollisionDetectorComp.o
[ 97%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/__/SequencePlayer/interpolator.o
[ 98%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/CollisionDetector.o
[ 98%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/CollisionDetectorService_impl.o
[ 98%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/VclipLinkPair.o
[ 99%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/__/__/lib/util/BVutil.o
[ 99%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/vclip_1.0/src/vclip.o
[100%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/vclip_1.0/src/PolyTree.o
[100%] Building CXX object rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/vclip_1.0/src/mv.o
Linking CXX executable ../../bin/CollisionDetectorComp
CMakeFiles/CollisionDetectorComp.dir/__/__/lib/util/BVutil.o: In function `convertToConvexHull(hrp::Link*)':
BVutil.cpp:(.text+0x564): undefined reference to `qh_new_qhull(int, int, double*, unsigned int, char*, _Filet*, _Filet*)'
BVutil.cpp:(.text+0x571): undefined reference to `qh_triangulate()'
BVutil.cpp:(.text+0x576): undefined reference to `qh_vertexneighbors()'
BVutil.cpp:(.text+0x60a): undefined reference to `qh_facet3vertex(facetT*)'
BVutil.cpp:(.text+0x647): undefined reference to `qh_pointid(double*)'
BVutil.cpp:(.text+0x6e9): undefined reference to `qh_freeqhull(unsigned int)'
BVutil.cpp:(.text+0x6fb): undefined reference to `qh_memfreeshort(int*, int*)'
BVutil.cpp:(.text+0x7c7): undefined reference to `qh_pointid(double*)'
cc: /usr/qnx650/host/qnx6/x86/usr/bin/ntox86-ld error 1
make[2]: *** [bin/CollisionDetectorComp] Error 1
make[1]: *** [rtc/CollisionDetector/CMakeFiles/CollisionDetectorComp.dir/all] Error 2
make: *** [all] Error 2
```

k-okada, 2016-09-06:
すっかり忘れているけど，libqp を link しているかな．
make VERBOSE=1 で link (ld)しているところを探そう．

pazeshun, 2016-09-06:
linkはできてます。
https://github.com/NIRALUser/ARCTIC/blob/master/itkEMS/qhull/qhull_interface.cpp#L8-L42 を読むと、 https://github.com/fkanehiro/hrpsys-base/blob/master/lib/util/BVutil.cpp#L2-L4 のように書くのは間違いみたいです。
そこで、BVutil.cppの冒頭を以下のように修正してみたところ、ビルドが通りました。
```
#include <iostream>
//extern "C" {
//#include <libqhull/qhull_a.h>
//}
#if defined(__cplusplus)
extern "C"
{
#endif
#include <stdio.h>
#include <stdlib.h>
#include <libqhull/libqhull.h>
#include <libqhull/mem.h>
#include <libqhull/qset.h>
#include <libqhull/geom.h>
#include <libqhull/merge.h>
#include <libqhull/poly.h>
#include <libqhull/io.h>
#include <libqhull/stat.h>
#if defined(__cplusplus)
}
#endif
#include <hrpModel/Link.h>
#include "BVutil.h"
```
これでinstallしてみて、動くかやってみます。

k-okada, 2016-09-06:
> qhull/qhull_interface.cpp#L8-L42 を読むと、 https://github.com/fkanehiro/
> hrpsys-base/blob/master/lib/util/BVutil.cpp#L2-L4 のように書くのは間違いみたいです。

実機がうごいたらでいいけど
In MVC とかいてあるから原因は違うと思う
qhull_a.hを見て見て

pazeshun, 2016-09-06:
In MVCとはどこのことでしょうか？

k-okada, 2016-09-06:
https://github.com/NIRALUser/ARCTIC/blob/master/itkEMS/qhull/qhull_interface.cpp#L10

pazeshun, 2016-09-06:
MVCのくだりを見落としていました。
ですが、私の記憶では、`extern "C"`をした場合に`math.h`内の関数でエラーが出るという症状がHiroでも発生して、そこで時間切れになったように思います。
なので、この解決策が使えると考えて実装してみた次第です。
なお、rtmros_hironx/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.pyがデフォルトの状態で手元PCからHiroに接続すると動きましたが、CollisionDetectorを入れて動かして見たところ、以下のようにエラーが出て、Hiroを動かせませんでした。
```
$ ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host hiro014
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with hiro014:15005
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7feb3c3405f0> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7feb3c3405f0> isActive? = False 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7feb3c340248>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7feb3c340560>
[hrpsys.py]    version  =  1.0
;;
;; Loading ImpedanceController < 315.1.9
;;
[hrpsys.py]   bodyinfo URL = file:///opt/jsk/etc/HIRONX/model/main.wrl
[hrpsys.py] no hrpsys components found running. creating now
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")
[hrpsys.py] create Comp -> SequencePlayer : <hrpsys.rtm.RTcomponent instance at 0x7feb3c352758> (1.0)
[hrpsys.py] create CompSvc -> SequencePlayer Service : <OpenHRP._objref_SequencePlayerService instance at 0x7feb3c354248>
[hrpsys.py]   eval : [self.sh, self.sh_svc, self.sh_version] = self.createComp("StateHolder","sh")
[hrpsys.py] create Comp -> StateHolder : <hrpsys.rtm.RTcomponent instance at 0x7feb3c353638> (1.0)
[hrpsys.py] create CompSvc -> StateHolder Service : <OpenHRP._objref_StateHolderService instance at 0x7feb3c36e5a8>
[hrpsys.py]   eval : [self.fk, self.fk_svc, self.fk_version] = self.createComp("ForwardKinematics","fk")
[hrpsys.py] create Comp -> ForwardKinematics : <hrpsys.rtm.RTcomponent instance at 0x7feb3c354320> (1.0)
[hrpsys.py] create CompSvc -> ForwardKinematics Service : <OpenHRP._objref_ForwardKinematicsService instance at 0x7feb3c34f4d0>
[hrpsys.py]   eval : [self.ic, self.ic_svc, self.ic_version] = self.createComp("ImpedanceController","ic")
[hrpsys.py] create Comp -> ImpedanceController : <hrpsys.rtm.RTcomponent instance at 0x7feb3c353488> (1.0)
[hrpsys.py] create CompSvc -> ImpedanceController Service : <OpenHRP._objref_ImpedanceControllerService instance at 0x7feb3c363518>
[hrpsys.py]   eval : [self.el, self.el_svc, self.el_version] = self.createComp("SoftErrorLimiter","el")
[hrpsys.py] create Comp -> SoftErrorLimiter : <hrpsys.rtm.RTcomponent instance at 0x7feb3c34f200> (1.0)
[hrpsys.py] create CompSvc -> SoftErrorLimiter Service : <OpenHRP._objref_SoftErrorLimiterService instance at 0x7feb3c352320>
[hrpsys.py]   eval : [self.co, self.co_svc, self.co_version] = self.createComp("CollisionDetector","co")
[hrpsys.py] create Comp -> CollisionDetector : <hrpsys.rtm.RTcomponent instance at 0x7feb3c35efc8> (1.0)
[hrpsys.py] create CompSvc -> CollisionDetector Service : <OpenHRP._objref_CollisionDetectorService instance at 0x7feb3c352908>
[hrpsys.py]   eval : [self.sc, self.sc_svc, self.sc_version] = self.createComp("ServoController","sc")
[hrpsys.py] create Comp -> ServoController : <hrpsys.rtm.RTcomponent instance at 0x7feb3c353bd8> (1.0)
[hrpsys.py] create CompSvc -> ServoController Service : <OpenHRP._objref_ServoControllerService instance at 0x7feb3c34ff80>
[hrpsys.py]   eval : [self.log, self.log_svc, self.log_version] = self.createComp("DataLogger","log")
[hrpsys.py] create Comp -> DataLogger : <hrpsys.rtm.RTcomponent instance at 0x7feb3c35e950> (1.0)
[hrpsys.py] create CompSvc -> DataLogger Service : <OpenHRP._objref_DataLoggerService instance at 0x7feb3c34fb48>
[hrpsys.py]   eval : [self.rmfo, self.rmfo_svc, self.rmfo_version] = self.createComp("AbsoluteForceSensor","rmfo")
[hrpsys.py] create Comp -> AbsoluteForceSensor : <hrpsys.rtm.RTcomponent instance at 0x7feb3c356bd8> (1.0)
[hrpsys.py] create CompSvc -> AbsoluteForceSensor Service : <OpenHRP._objref_AbsoluteForceSensorService instance at 0x7feb3c363bd8>
[hrpsys.py] connecting hrpsys components
[rtm.py]    Connect sh.qOut - ic.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect ic.q - co.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect co.q - el.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect el.q - RobotHardware0.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect RobotHardware0.servoState - el.servoStateIn (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect RobotHardware0.q - sh.currentQIn (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[... cut ...]
-> 1650         return self.rh_svc.getStatus()
   1651 
   1652     def isCalibDone(self):

/opt/ros/indigo/lib/python2.7/dist-packages/hrpsys/RobotHardwareService_idl.py in getStatus(self, *args)
    231 
    232     def getStatus(self, *args):
--> 233         return _omnipy.invoke(self, "getStatus", _0_OpenHRP.RobotHardwareService._d_getStatus, args)
    234 
    235     def power(self, *args):

OBJECT_NOT_EXIST: CORBA.OBJECT_NOT_EXIST(omniORB.OBJECT_NOT_EXIST_NoMatch, CORBA.COMPLETED_NO)
```
いろいろ忘れているので、思い出しながら作業します。

pazeshun, 2016-09-06:
CollisionDetector入れなかった時は以下のようなログになります。
```
$ ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host hiro014
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
configuration ORB with hiro014:15005
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efc55f0> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efc55f0> isActive? = False 
[hrpsys.py] simulation_mode : False
[hrpsys.py]  Hrpsys controller version info: 
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7f1e0efc5248>
[hrpsys.py]    ref =  <RTM._objref_Manager instance at 0x7f1e0efc5560>
[hrpsys.py]    version  =  1.0
;;
;; Loading ImpedanceController < 315.1.9
;;
[hrpsys.py]   bodyinfo URL = file:///opt/jsk/etc/HIRONX/model/main.wrl
[hrpsys.py] no hrpsys components found running. creating now
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")
[hrpsys.py] create Comp -> SequencePlayer : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efd7950> (1.0)
[hrpsys.py] create CompSvc -> SequencePlayer Service : <OpenHRP._objref_SequencePlayerService instance at 0x7f1e0efd9248>
[hrpsys.py]   eval : [self.sh, self.sh_svc, self.sh_version] = self.createComp("StateHolder","sh")
[hrpsys.py] create Comp -> StateHolder : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efd88c0> (1.0)
[hrpsys.py] create CompSvc -> StateHolder Service : <OpenHRP._objref_StateHolderService instance at 0x7f1e0eff55a8>
[hrpsys.py]   eval : [self.fk, self.fk_svc, self.fk_version] = self.createComp("ForwardKinematics","fk")
[hrpsys.py] create Comp -> ForwardKinematics : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efd9368> (1.0)
[hrpsys.py] create CompSvc -> ForwardKinematics Service : <OpenHRP._objref_ForwardKinematicsService instance at 0x7f1e0efd44d0>
[hrpsys.py]   eval : [self.ic, self.ic_svc, self.ic_version] = self.createComp("ImpedanceController","ic")
[hrpsys.py] create Comp -> ImpedanceController : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efd8c68> (1.0)
[hrpsys.py] create CompSvc -> ImpedanceController Service : <OpenHRP._objref_ImpedanceControllerService instance at 0x7f1e0efe8518>
[hrpsys.py]   eval : [self.el, self.el_svc, self.el_version] = self.createComp("SoftErrorLimiter","el")
[hrpsys.py] create Comp -> SoftErrorLimiter : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efd4908> (1.0)
[hrpsys.py] create CompSvc -> SoftErrorLimiter Service : <OpenHRP._objref_SoftErrorLimiterService instance at 0x7f1e0efd7320>
[hrpsys.py]   eval : [self.sc, self.sc_svc, self.sc_version] = self.createComp("ServoController","sc")
[hrpsys.py] create Comp -> ServoController : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efe3560> (1.0)
[hrpsys.py] create CompSvc -> ServoController Service : <OpenHRP._objref_ServoControllerService instance at 0x7f1e0efd98c0>
[hrpsys.py]   eval : [self.log, self.log_svc, self.log_version] = self.createComp("DataLogger","log")
[hrpsys.py] create Comp -> DataLogger : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efd8b48> (1.0)
[hrpsys.py] create CompSvc -> DataLogger Service : <OpenHRP._objref_DataLoggerService instance at 0x7f1e0efdbe18>
[hrpsys.py]   eval : [self.rmfo, self.rmfo_svc, self.rmfo_version] = self.createComp("AbsoluteForceSensor","rmfo")
[hrpsys.py] create Comp -> AbsoluteForceSensor : <hrpsys.rtm.RTcomponent instance at 0x7f1e0efd40e0> (1.0)
[hrpsys.py] create CompSvc -> AbsoluteForceSensor Service : <OpenHRP._objref_AbsoluteForceSensorService instance at 0x7f1e0efe8d40>
[hrpsys.py] connecting hrpsys components
[rtm.py]    Connect sh.qOut - ic.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect ic.q - el.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect el.q - RobotHardware0.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect RobotHardware0.servoState - el.servoStateIn (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect RobotHardware0.q - sh.currentQIn (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect RobotHardware0.q - fk.q (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect sh.qOut - fk.qRef (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect seq.qRef - sh.qIn (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect seq.tqRef - sh.tqIn (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[rtm.py]    Connect seq.basePos - sh.basePosIn (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[... cut ...]
/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/ros_client.pyc in __init__(self, jointgroups)
     84         except socket.error as e:
     85             errormsg = '[ros_client] ros master is not running, so do not create ros client...'
---> 86             raise ROSInitException(errormsg)
     87         except Exception as e:
     88             errormsg = '[ros_client] Unknown exception occurred, so do not create ros client...'

ROSInitException: [ros_client] ros master is not running, so do not create ros client...
```

k-okada, 2016-09-06:
なるほど
ということは先にmath.hをincludeすればよさそうだね

エラーはqnx側の/opt/jsk/var/log/rtcd.log
みたいなところを見よう

2016年9月6日火曜日、pazeshun<notifications@github.com

pazeshun, 2016-09-06:
明日math.hを先にロードしてみるのと、ログを見てみます

emijah, 2016-09-07:
KRCの齋藤元です。

math.hで起きたエラーを見せていただくことはできますか？

pazeshun, 2016-09-07:
@emijah 齋藤様
遅くなって申し訳ありません。以下がエラーの全文になります。
https://drive.google.com/file/d/0BxxBA3J-CunGZGJIMGFkNmpHNTQ/view?usp=sharing
C++で記述されたものをCで解釈しきれずに起こるエラーであると思います。

pazeshun, 2016-09-07:
@k-okada 
BVutil.cppの冒頭を以下のようにすることでもエラーを回避出来ました。
```
#include <iostream>
#include <math.h>
extern "C" {
#include <libqhull/qhull_a.h>
}
```

pazeshun, 2016-09-07:
通信失敗した時のログは以下のようになっています。
```
$ cat /opt/jsk/var/log/rtcd.log 
Logger::Logger: streambuf address = 0x805fc70
hrpExecutionContext is registered
pdgains.file_name: /opt/jsk/etc/HIRONX/hrprtc/PDgains.sav
dof = 15
client: msg_no: _IO_MAX + 4
client: msg_reply:
rom # 4, soft # 300, 28/9/2011
serial # 5171, model # 129, cal day 28/9/2011
Newton, Newton*meter*10, mm*10, 14 bits, 255 ch
(Crom # 4, soft # 300, 4/10/2011
serial # 5172, model # 129, cal day 4/10/2011
Newton, Newton*meter*10, mm*10, 14 bits, 255 ch
(C
dummy IOB is opened
the number of gyros = 0
the number of accelerometers = 0
the number of force sensors = 2
lhsensor
rhsensor
dummy IOB is opened
period = 5[ms], priority = 49
```

k-okada, 2016-09-07:
CollisionDetector入れる／入れない．は .so を置く/置かない，という切り替え方という理解でいい？

/opt/jsk/etc/HIRONX/hrprtc/rtcdRobotMode.confに
example.CollisionDetector.config_file: /opt/jsk/etc/HIROINX/hrprtc/Robot.conf
みたいなのはあるかな．
RTCTREE_NAMESERVERS=hiro014:15005 rtls  hiro014:15005
か
RTCTREE_NAMESERVERS=hiro014:2809 rtls hiro014:2809
はどうなっているかな
そこで，co.rtcがみえていれば，
RTCTREE_NAMESERVERS=hiro014:15005 rtcat -l  hiro014:15005/co.rtc
かな．

pazeshun, 2016-09-07:
> CollisionDetector入れる／入れない．は .so を置く/置かない，という切り替え方という理解でいい？

そうではなく、手元PCのhironx_ros_bridgeパッケージ内のhironx_client.pyのrtclistのCollisionDetectorをコメントアウトするかしないかという切り替え方です。

k-okada, 2016-09-07:
なるほど，それで良いです．で，
```
/opt/ros/indigo/share/hironx_ros_bridge/scripts/hironx.py in <module>()
     78     # ROS Client
     79     try:
---> 80         ros = ROS_Client()
```
の` ros = ROS_Client() `は一旦コメントアウトして進めるのがいいね．

pazeshun, 2016-09-07:
> /opt/jsk/etc/HIRONX/hrprtc/rtcdRobotMode.confに
> example.CollisionDetector.config_file: /opt/jsk/etc/HIROINX/hrprtc/Robot.conf
> みたいなのはあるかな．

これはありました。

k-okada, 2016-09-07:
```
[rtm.py]    Failed to connect None to [None]([None])
[rtm.py]    Failed to connect None to [None]([None])
[rtm.py]    Failed to connect None to [None]([None])
[rtm.py]    Failed to connect None to [None]([None])
[rtm.py]    Failed to connect RobotHardware0.servoState to None([None])
```
あたりのエラーが
https://github.com/fkanehiro/hrpsys-base/blob/315.9.0/python/hrpsys_config.py#L284-L532
のどこから出ているか調べたい．

pazeshun, 2016-09-07:
> RTCTREE_NAMESERVERS=hiro014:15005 rtls hiro014:15005
> か
> RTCTREE_NAMESERVERS=hiro014:2809 rtls hiro014:2809
> はどうなっているかな

すみません、これの意味がわからないのですが、以下のような操作であってますでしょうか？
```
pazeshun@pazeshun-ThinkPad-T460s:~$ export RTCTREE_NAMESERVERS=hiro014:2809
pazeshun@pazeshun-ThinkPad-T460s:~$ rtls hiro014:2809
rtls: Cannot list ports.
pazeshun@pazeshun-ThinkPad-T460s:~$ export RTCTREE_NAMESERVERS=hiro014:15005
pazeshun@pazeshun-ThinkPad-T460s:~$ rtls hiro014:15005
rtls: Cannot list ports.
```

k-okada, 2016-09-07:
はい．
```
pazeshun@pazeshun-ThinkPad-T460s:~$ export RTCTREE_NAMESERVERS=hiro014:15005
pazeshun@pazeshun-ThinkPad-T460s:~$ rtls hiro014:15005/
```
だ．最後に/がひつよう．
export するとその後のshellで，その環境変数が有効
```
RTCTREE_NAMESERVERS=hiro014:15005 rtls hiro014:15005
```
すると，この行だけ環境変数が有効．

pazeshun, 2016-09-07:
```
$ export RTCTREE_NAMESERVERS=hiro014:15005
$ rtls hiro014:15005/
sc.rtc  ModelLoader  el.rtc   hiro014.host_cxt/  fk.rtc    seq.rtc
sh.rtc  log.rtc      *co.rtc  ic.rtc            rmfo.rtc  RobotHardware0.rtc
```
co.rtcには*がついてます。

k-okada, 2016-09-07:
ソースのバージョンは315.1.10 かな？co.rtcができていて＊はエラーかな．rtcat -l で分かるはず．
https://github.com/fkanehiro/hrpsys-base/blob/315.1.10/rtc/CollisionDetector/CollisionDetector.cpp
みると，
co: onInitialize()
は表示されているはずなんだけどなぁ．

pazeshun, 2016-09-07:
すみません、私の一つ前のコメントは、CollisionDetectorを入れなかった場合のものでした。
入れると、前に上げたエラーがipython側で出るのと、
```
$ rtls hiro014:15005/
*sc.rtc  ModelLoader  *el.rtc           *co.rtc  *fk.rtc    *seq.rtc
*sh.rtc  *log.rtc     hiro014.host_cxt/  *ic.rtc  *rmfo.rtc  RobotHardware0.rtc
```
のようになります。

pazeshun, 2016-09-07:
```
$ rtcat -l hiro014:15005/co.rtc
rtcat: Zombie object: hiro014:15005/co.rtc
```
のようです。

k-okada, 2016-09-07:
お，ぞんびだとアレだな．どうやってrtcd再起動している？ slay rtcdみたいな？
NameServerをリスタートしないとゾンビは生きていそうなので，リブートかomniNamesをslayして，
NameServer.sh を立ち上げ直すか

pazeshun, 2016-09-07:
> ソースのバージョンは315.1.10 かな？

多分、ソースの元のバージョンは315.1.7で、それに、OpenCVがインストールされていなくてもCollisionDetectorを使えるようにするパッチ（https://patch-diff.githubusercontent.com/raw/fkanehiro/hrpsys-base/pull/428.diff ）を若干編集して当てたものです。
今、現在のhrpsys-baseとのdiffをとってみようかと思って、hrpsys-baseをクローンしています。

k-okada, 2016-09-07:
package.xml か CHANGELOG.rst みたらバージョンはわかるともう．

pazeshun, 2016-09-07:
> お，ぞんびだとアレだな．どうやってrtcd再起動している？ slay rtcdみたいな？
> NameServerをリスタートしないとゾンビは生きていそうなので，リブートかomniNamesをslayして，
> NameServer.sh を立ち上げ直すか

とりあえず再起動してました。

[23 more replies]