start-jsk/rtmros_hironx #435: hironx.py connection inter-host; RobotHardware wasn't found

130s, 2016-02-06:
Host1: Running `rtmlaunch hironx_ros_bridge hironx_ros_bridge_simulation.launch`
Host2: "ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py -- --host HOST1_NAME"

`RobotHardware` wasn't found error happened. These 2 hosts are in the same LAN.

k-okada, 2016-02-06:
--host must be target to the RTM name server name, not HOST1_NAME,
if you wan to connect ROS client in hironx.py to ROS master running at
HOST1_NAME,
use ROS_MASTER_URI as always.

130s, 2016-02-06:
I've updated my original post for clarification. I don't think I used a wrong RTM host name.

`ping` worked between the hosts.

k-okada, 2016-02-07:
Oh, you're using simulation ,
In that case you have to manually specify `--robot HiroNX(Robot)0` to override 
https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_ros_bridge/scripts/hironx.py#L66

130s, 2016-02-08:
Thanks you for the clarification.

[Added to wiki](http://wiki.ros.org/rtmros_nextage/Tutorials/Run%20simulation#Connecting_to_a_simulated_robot_running_on_a_remote_host) as well.

closed 2016-02-08 by 130s