start-jsk/rtmros_hironx #447: Issue with tutorial Programming_Hiro_NEXTAGE_OPEN_ROS

VinceDT, 2016-03-02:
Hey guys,

I am working through the tutorials and at the Programming_Hiro_NEXTAGE_OPEN_ROS tutorial. 
Having started:
`rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch`
`roslaunch nextage_moveit_config moveit_planning_execution.launch`
```ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py```
I get:

```
In [19]: ros.go_init()
[INFO] [WallTime: 1456932882.932612] [3330.960000] *** go_init begins ***
---------------------------------------------------------------------------
MoveItCommanderException                  Traceback (most recent call last)
<ipython-input-19-dd5d44413d43> in <module>()
----> 1 ros.go_init()

/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/ros_client.pyc in go_init(self, init_pose_type, task_duration)
    204         elif 1 == init_pose_type:
    205             posetype_str = 'init_rtm_factory'
--> 206         self.MG_BOTHARMS.set_named_target(posetype_str)
    207         self.MG_BOTHARMS.go()
    208 

/opt/ros/indigo/lib/python2.7/dist-packages/moveit_commander/move_group.pyc in set_named_target(self, name)
    280         """ Set a joint configuration by name. The name can be a name previlusy remembered with remember_joint_values() or a configuration specified in the SRDF. """
    281         if not self._g.set_named_target(name):
--> 282             raise MoveItCommanderException("Unable to set target %s. Is the target within bounds?" % name)
    283 
    284     def remember_joint_values(self, name, values = None):

MoveItCommanderException: Unable to set target init_rtm. Is the target within bounds?
```

Using:
Ubuntu 14.04 up-to-date
ROS indigo up-to-date
hironx packages installed yesterday via apt

Additionally there are errors at startup (maybe relevant to the problem):

rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch:

`[ERROR] [1456933341.552714727]: Target Node visual1/node_joint0_axis0 NOT found!!!`
`[ERROR] [1456933344.719268271]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?`
`[hrpsys_py-4] process has died ...`

roslaunch nextage_moveit_config moveit_planning_execution.launch:
```
[ERROR] [1456933569.973900787]: Group 'torso' is not a chain
[ERROR] [1456933569.973948958]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'torso'
[ERROR] [1456933569.974412217]: Kinematics solver could not be instantiated for joint group torso.
```

thanks!

130s, 2016-03-02:
Hm, I can't reproduce this error. Can you do us a favor for things?

- give the result of this command (mine is included as a reference):
 ```
$ dpkg -p ros-indigo-hironx-moveit-config ros-indigo-hironx-ros-bridge ros-indigo-moveit-commander |grep Ver
Version: 1.1.11-0trusty-20160218-195453-0800
Version: 1.1.11-0trusty-20160218-193852-0800
Version: 0.6.0-0trusty-20160205-202620-0800
```

- What happens with `ros.go_init(init_pose_type=1)`?

- Result of `rosnode list`?

130s, 2016-03-02:
I just noticed you're actually willing to run Nextage. So replace the dpkg command with this:

 ```
$ dpkg -p ros-indigo-hironx-moveit-config ros-indigo-hironx-ros-bridge ros-indigo-nextage-moveit-config ros-indigo-nextage-ros-bridge ros-indigo-nextage-description ros-indigo-moveit-commander |grep Ver
Version: 1.1.11-0trusty-20160218-195453-0800
Version: 1.1.11-0trusty-20160218-193852-0800
Version: 0.7.6-0trusty-20160218-201252-0800
Version: 0.7.6-0trusty-20160218-195737-0800
Version: 0.7.6-0trusty-20160210-125241-0800
Version: 0.6.0-0trusty-20160205-202620-0800
```

VinceDT, 2016-03-03:
Hey,

- Versions:

```
dpkg -p ros-indigo-hironx-moveit-config ros-indigo-hironx-ros-bridge ros-indigo-nextage-moveit-config ros-indigo-nextage-ros-bridge ros-indigo-nextage-description ros-indigo-moveit-commander |grep Ver
Version: 1.1.7-0trusty-20160208-142609-0800
Version: 1.1.7-0trusty-20160208-134735-0800
Version: 0.7.5-0trusty-20160208-150117-0800
Version: 0.7.5-0trusty-20160208-142556-0800
Version: 0.7.5-0trusty-20160127-152705-0800
Version: 0.6.0-0trusty-20160205-202620-0800
```
- `ros.go_init(init_pose_type=1)` 
```
In [1]: ros.go_init(init_pose_type=1)
[INFO] [WallTime: 1456994658.843591] [230.885000] *** go_init begins ***
---------------------------------------------------------------------------
MoveItCommanderException                  Traceback (most recent call last)
<ipython-input-1-92671653e45b> in <module>()
----> 1 ros.go_init(init_pose_type=1)

/opt/ros/indigo/lib/python2.7/dist-packages/hironx_ros_bridge/ros_client.pyc in go_init(self, init_pose_type, task_duration)
    204         elif 1 == init_pose_type:
    205             posetype_str = 'init_rtm_factory'
--> 206         self.MG_BOTHARMS.set_named_target(posetype_str)
    207         self.MG_BOTHARMS.go()
    208 

/opt/ros/indigo/lib/python2.7/dist-packages/moveit_commander/move_group.pyc in set_named_target(self, name)
    280         """ Set a joint configuration by name. The name can be a name previlusy remembered with remember_joint_values() or a configuration specified in the SRDF. """
    281         if not self._g.set_named_target(name):
--> 282             raise MoveItCommanderException("Unable to set target %s. Is the target within bounds?" % name)
    283 
    284     def remember_joint_values(self, name, values = None):

MoveItCommanderException: Unable to set target init_rtm_factory. Is the target within bounds?
``` 
- rosnode list 
```
/DataLoggerServiceROSBridge
/ForwardKinematicsServiceROSBridge
/HrpsysJointTrajectoryBridge
/HrpsysSeqStateROSBridge
/SequencePlayerServiceROSBridge
/StateHolderServiceROSBridge
/collision_state
/diagnostic_aggregator
/hironx_ros_client
/hrpsys_profile
/hrpsys_ros_diagnostics
/hrpsys_state_publisher
/move_group
/moveit_python_wrappers_1456994431345037205
/moveit_python_wrappers_1456994634747139003
/rosout
/rviz_md1hugwc_29247_1699392634724605031
 ```

Same error on the machine of a colleague.

thanks for the help!

k-okada, 2016-05-02:
@VinceDT do you still have this error ?

130s, 2017-05-22:
Closing due to inactivity. Feel free to reopen with updated info if the issue persists.

closed 2017-05-22 by 130s