start-jsk/rtmros_hironx #453: Redundant "connectPorts(self.rh.port(sensor), self.ic.port(sensor))"?

130s, 2016-04-12:
@k-okada @hyaguchijsk 

[In HIRONX](https://github.com/start-jsk/rtmros_hironx/blob/afe78e916ce0f8125eea0a86e818ca31fce29d6e/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L177-L181), when IC is present (along with other conditions), RH connects to IC port.

```
 def init(self, robotname="HiroNX(Robot)0", url=""):
:
        # connect ic if needed
        for sensor in ['lhsensor' , 'rhsensor']:
            if self.ic and self.ic.port(sensor) and self.ic.port(sensor).get_port_profile() and \
                    not self.ic.port(sensor).get_port_profile().connector_profiles :
                connectPorts(self.rh.port(sensor), self.ic.port(sensor))
```

However, it seems already done [in its parent class, `HrpsysConfigurator`](https://github.com/fkanehiro/hrpsys-base/blob/872b832ba658618c015189fdded4afe3f749a0e7/python/hrpsys_config.py#L440-L443):
```
 def connectComps(self):
:
        elif self.ic: # if the robot does not have rmfo and kf, but have ic
            for sen in filter(lambda x: x.type == "Force", self.sensors):
                connectPorts(self.rh.port(sen.name),
                             self.ic.port(sen.name))
```

Why do we need the one in HIRONX? Thank you in advance.

k-okada, 2016-04-15:
sorry, I do not remember whey I added this code, but since these two change
has been made same day, i think I put the code to both hironx and hrpsys
and use hironx one until hrpsys has been released. and forget to remove
hironx one when hrpsys has released.
anyway i think we can remove hironx settings

130s, 2016-04-15:
Thank you for the comment. #454 is opened and I'll ask a robot owner who has f/t sensors for testing.

closed 2016-04-15 by 130s