start-jsk/rtmros_hironx #474: Incoherent joint limits

miguelprada, 2016-10-21:
I just realised that the joint limits in `hironx_moveit_config` package are not coherent between [`HiroNX.urdf`](https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_moveit_config/models/HiroNX.urdf#L500) and [`joint_limits.yaml`](https://github.com/start-jsk/rtmros_hironx/blob/indigo-devel/hironx_moveit_config/config/joint_limits.yaml#L4).

I'm working with the assumption that the URDF specifies hardware limits, and the MoveIt-specific `joint_limits.yaml` file can be used to override these values for planning. If that's correct, it makes no sense to plan with velocity limits higher those really supported by the robot hardware.

Is my understanding of how this works incorrect? Are there any _real_ velocity limits that could be introduced in the URDF?

130s, 2016-10-21:
Quick response/guess is that some information in the URDF in question was made up for certain reason at the time of its creation (just a guess...it might have been more crucial for the manufacturer then, or not available for any reason etc.) so that it's possible that all the info is not necessarily correct.

7675t, 2018-02-13:
Related to this, I cannot find which file should be modified to change the velocity limit of joints. 
Could you point me which file is actually used with HiroNX ?