start-jsk/rtmros_hironx #480: Stop running ServoController RTC by default?

130s, 2016-11-14:
Hand commands available in hironx client (e.g. [HandOpen](https://github.com/start-jsk/rtmros_hironx/blob/f523a54e9cd96af54a29d5613730b91600eb95a1/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L433)) can be confusing to the users of both Hironx without servo-based hands and NXO that don't come with servo-hands (e.g. https://github.com/tork-a/rtmros_nextage/issues/274).
Changes needed I can think of:
- Move Hand* methods from [hironx_client.py](https://github.com/start-jsk/rtmros_hironx/blob/f523a54e9cd96af54a29d5613730b91600eb95a1/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py) to another class. Users who need them need to be able to use those (but who are they?).
- Remove SC from the config stored internally in the robot.

130s, 2017-04-05:
Testing to disable Servo Controller RTC by commenting out the following line in `/opt/jsk/etc/HIRONX/hrprtc/rtcdRobotMode.conf`:

```
:
## Grasper
example.ServoController.config_file: /opt/jsk/etc/HIRONX/hrprtc/Robot.conf
:
```

130s, 2017-04-05:
Rebooting the robot after https://github.com/start-jsk/rtmros_hironx/issues/480#issuecomment-291769946, the ServoController seems not running as expected (no `sc.rtc` found).

```
$ rtls nextage:15005/
ModelLoader       el.rtc              fk.rtc  seq.rtc  sh.rtc  log.rtc  ic.rtc  rmfo.rtc
nextage.host_cxt/  RobotHardware0.rtc
```

k-okada, 2017-04-07:
> Rebooting the robot after #480 (comment)

hironx comes with servo hand by default ( I think), so if nextage do not
have servo hand by default, Change/Overwrite them on nextage_client.py