start-jsk/rtmros_hironx #524: Maker ArrayのStatusがエラーになる

aislab, 2018-02-13:
HiroNXで画像のようにエラーになります。
パッケージは、Indigoリリースです。
WAIST_Linkというリンクはロボットは持っていないようなのですが、どこを変更すればこのエラーが消えるでしょうか。

[image]

7675t, 2018-02-14:
このMarkerArrayは，両腕の間に出てくる緑の線分状のマーカだと思います．手元でシミュレータで試すと，エラーも表示されずマーカは表示されます．実機で起動した際に，このエラーが表示されました．

/collistion_detector_marker_arrayを発行しているのは，collision_stateノードで，これは，hrpsys_ros_bridge/launch/collision_detector.launchで起動されています．

現地で記録しておけばよかったのですが，直接の原因は/collision_detector_marker_arrayのframe_idが"WAIST_Link"になっているためだと思います．ロボットモデルには"WAIST_Link"というリンクは存在しないのではと思います．

実機でこのエラーが出る時の情報を教えてもらえないでしょうか ＞ @aislab 
何が必要かよく分かりませんが，とりあえず，

```
$ rostopic echo /collision_detector_marker_array 
$ rosparam get /robot_description
```
の結果あたりを．

あと一応，ワークスペースが使われていないこと（roscd したら/optに行くこと）をもう一度確認してみてください．

aislab, 2018-02-15:
```
$ rostopic echo /collision_detector_marker_array 
$ rosparam get /robot_description
```
の実行結果です．
robot_descriptionについては数値のみの部分は省略しています．

<details>
  <summary>collision_detector_marker_array</summary>

```
  markers: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "WAIST_Link"
    ns: ''
    id: 0
    type: 5
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.01
      y: 0.0
      z: 0.0
    color: 
      r: 0.0
      g: 1.0
      b: 0.0
      a: 1.0
    lifetime: 
      secs: 1
      nsecs:         0
    frame_locked: False
    points: 
      - 
        x: 0.00430564472553
        y: -0.124977923186
        z: 0.084016598715
      - 
        x: -0.000327781550538
        y: 0.123422017915
        z: 0.0767881214276
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "WAIST_Link"
    ns: ''
    id: 1
    type: 2
    action: 0
[... 2543 lines cut ...]
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid2.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint3_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid3.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid3.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint4_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid4.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid4.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint5_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid5.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid5.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint6_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid6.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid6.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint7_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid7.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid7.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint8_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid8.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid8.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint9_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid9.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<value>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid9.axis0_value</param>\n\
  \t\t\t\t</value>\n\t\t\t</bind_joint_axis>\n\t\t\t<bind_joint_axis target=\"visual1/node_joint10_axis0\"\
  >\n\t\t\t\t<axis>\n\t\t\t\t\t<param>kscene_kmodel1_inst_robot1_kinematics_kmodel1_inst_jointsid10.axis0</param>\n\
  \t\t\t\t</axis>\n\t\t\t\t<val

7675t, 2018-02-15:
やはり，HiroNXのモデルと，collision_statesノードの想定しているWAISTリンクの名前が違うのが問題のようです．HiroNXのモデルファイルは，ロボットの体内のQNX側にありそちらを変更するのは難しいので，とりあえず，

```
rosrun tf static_transform_publisher 0 0 0 0 0 0 /WAIST /WAIST_Link 1
```

とWAIST_Link座標系を追加すると，エラーがなくなるのではないかと思うので，試してみてもらえないでしょうか．

もしこれで上手く行くようでしたら，ややアドホックですが，これをhironx_ros_bridge_real.launchに入れると良いです．その場合PRを出してもらえると助かります．

aislab, 2018-02-19:
実機環境にて
`rosrun tf static_transform_publisher 0 0 0 0 0 0 /WAIST /WAIST_Link 1`
を実行するとエラーは消えました．

ありがとうございます．

[image]

closed 2018-02-21 by aislab