start-jsk/rtmros_hironx #539: hironx.py fails due to CORBA.MARSHAL error on melodic

pazeshun, 2020-04-12:
On melodic, when I did
```
$ rtmlaunch hironx_ros_bridge hironx_ros_bridge_simulation.launch  # Tried multiple times to avoid https://github.com/start-jsk/rtmros_hironx/issues/538
$ ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py
```
, I got the following error:
```
Python 2.7.17 (default, Nov  7 2019, 10:07:09)
Type "copyright", "credits" or "license" for more information.                                                                       
                                                                                                                                     
IPython 5.5.0 -- An enhanced Interactive Python.                                                                                     
?         -> Introduction and overview of IPython's features.                                                                        
%quickref -> Quick reference.                                                                                                        
help      -> Python's own help system.                                                                                               
object?   -> Details about 'object', use 'object??' for extra details.                                                               
[rtm.py] Failed to parse corba.nameservers, use 46405e9dcc2a as nshost                                                               
[rtm.py] Failed to parse corba.nameservers, use 15005 as nsport                                                                      
[rtm.py] Failed to parse corba.master_manager, use 2810                                                                              
[rtm.py] configuration ORB with 46405e9dcc2a:15005                                                                                   
[rtm.py] configuration RTCManager with 46405e9dcc2a:2810                                                                             
[hrpsys.py] waiting ModelLoader                                                                                                      
[hrpsys.py] start hrpsys                                                                                                             
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py]  waitForRTCManagerAndRoboHardware has renamed to waitForRTCManagerAndRoboHardware: Please update your code
[rtm.py] trying to findRTCManager on port2810
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for HiroNX(Robot)0 : <hrpsys.rtm.RTcomponent instance at 0x7f9c6c015820> ( timeout 0 < 10)
[hrpsys.py] findComps -> HiroNX(Robot)0 : <hrpsys.rtm.RTcomponent instance at 0x7f9c6c015820> isActive? = True
[hrpsys.py] simulation_mode : True
[hrpsys.py]  Hrpsys controller version info:
[hrpsys.py]    ms =  <hrpsys.rtm.RTCmanager instance at 0x7f9c6c015690>
[hrpsys.py]    ref =  <RTM._objref_Manager object at 0x7f9c6bfeeed0>
------------------------------------------------------------------
MARSHAL                          Traceback (most recent call last)
/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_hironx/hironx_ros_bridge/scripts/hironx.py in <module>()
     80         args.modelfile = unknown[1]
     81     robot = hiro = hironx_client.HIRONX()
---> 82     robot.init(robotname=args.robot, url=args.modelfile)
     83
     84     # ROS Client

/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_hironx/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.pyc in init(self, robotname, url)
    343         if self.ms and self.ms.ref :
    344             print self.configurator_name, "  ref = ", self.ms.ref
--> 345         if self.ms and self.ms.ref and len(self.ms.ref.get_component_profiles()) > 0:
    346             print self.configurator_name, "  version  = ", self.ms.ref.get_component_profiles()[0].version
    347         if self.ms and self.ms.ref and len(self.ms.ref.get_component_profiles()) > 0 and StrictVersion(self.ms.ref.get_component_profiles()[0].version) < StrictVersion('315.2.0'):

/opt/ros/melodic/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/Manager_idl.py in get_component_profiles(self, *args)
    176
    177     def get_component_profiles(self, *args):
--> 178         return self._obj.invoke("get_component_profiles", _0_RTM.Manager._d_get_component_profiles, args)
    179
    180     def get_profile(self, *args):

MARSHAL: CORBA.MARSHAL(omniORB.MARSHAL_MessageSizeExceedLimitOnClient, CORBA.COMPLETED_YES)

In [1]: 
```

pazeshun, 2020-04-12:
I cound avoid this by setting `ORBgiopMaxMsgSize` explicitly like
```
ORBgiopMaxMsgSize=2147483648 ipython -i `rospack find hironx_ros_bridge`/scripts/hironx.py
```
However, I couldn't understand why this setting is required from melodic, not in indigo and kinetic.
I compared /etc/omniORB.cfg of Ubuntu 16.04 and Ubuntu 18.04, but both have the same setting:
```
giopMaxMsgSize = 2097152    # 2 MBytes.
```

k-okada, 2020-04-21:
fixed in https://github.com/start-jsk/rtmros_hironx/pull/540

これ，なんかおかしいですね．今まではopenhrp-model-loader を立ち上げるときに

https://github.com/start-jsk/rtmros_common/blob/ad3d58ccc8cb75398bdb9f994334c5dafc2e665e/hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launch#L59

の`  <env name="ORBgiopMaxMsgSize" value="2147483648" />` で起ち上げていて，
`hironx_clinet.py` からモデルを読み込むときに，この値を見てくれなくて，別途hironx_clinet.pyを立ち上げるときにも`giopMaxMsgSize` をセットしないといけない．

最初はこの`openhrp-model-loader` を立ち上げるときの`<env` が有効でないのかと思って，` ~/.ros/log/latest/roslaunch-*.log` を見ると，`u'ORBgiopMaxMsgSize': u'2147483648',`となっていて，これはユニコードの闇か？ともって，roslaunchを直して見たんだけどダメで，`/etc/omniORB.cfg ` を直接いじって，`openhrp-model-loader`  のときに`giopMaxMsgSize = 2147483648` で，`hironx_client.py`のときはもとに戻してもエラーになるから，もしかしたらomniorbの4.1->4.2で何か変わったかもしれない．

で，この問題hrpsys側で直したらいいのかもしれないけど（https://github.com/fkanehiro/hrpsys-base/pull/1289）とありえあず，https://github.com/start-jsk/rtmros_hironx/pull/540 で直しました．

`hrpsyspy` だけだとモデルは読まないんだけど，結構モデルを呼ぶ初期化ファイルもありそうで，そう言えばそういう報告なかったっけ？ @naoki-hiraoka @ishigurojsk @YoheiKakiuchi

ishiguroJSK, 2020-04-21:
jaxon_red_choreonoid.launchとかのいかにも通信量多そうなケースで，ORBgiopMaxMsgSizeの拡張しないとlaunch起動に失敗して，launch内の<env>とか.bashrcとか/etc/omniORB.cfgとかで効いたり効かなかったりする話は前からありますが，
launch起動が落ちるという現象しか見ていなかったので，それが**hrpsys_config.pyかとか，ModelLoaderなのかとかは把握していないです
最近は.bashrcに`export ORBgiopMaxMsgSize=2097152000`と書いておくととりあえずエラーが出なくなるので安心してしまっていました

YoheiKakiuchi, 2020-04-21:
jaxonはメッシュのファイルサイズ大きくて初期設定のままだとmodel-loaderで確実に落ちていたはずです。
メッシュが軽いと、なんとなく動き続けることもあるかもしれないですね。

etc とか bashrc とかなにもいじらない環境（例えばdocker環境）で16.04までは問題なかったので、
@k-okadaの推測が正しい気がします。
omniの問題か、launchの環境変数渡しの問題かということですね。

@pazeshun 
プロセスの環境変数が見えるようなので、これで落ちるプロセスの環境変数が適切か見てれるかな > 
https://qiita.com/suzutsuki0220/items/dbbe94963123b514e516

僕の知っている最少構成で落とせるのは以下になります。
```
rosrun openhrp3 start_omninames.sh 9999 &
sleep 1
ORBgiopMaxMsgSize=2147483648 openhrp-model-loader -ORBInitRef NameService=corbaloc:iiop:localhost:9999/NameService &
sleep 1
ORBgiopMaxMsgSize=2147483648 hrpsys-viewer $1 -ORBInitRef NameService=corbaloc:iiop:localhost:9999/NameService

$1のところにjaxonのwrlファイル
```

Naoki-Hiraoka, 2020-04-21:
Ubuntu16.04のHRP2では，launchに`ORBgiopMaxMsgSize`を記入しておけば，別端末からipythonで立ち上げた`config.py`も含めて落ちることはありませんでした．

pazeshun, 2020-04-21:
@YoheiKakiuchi 
> @pazeshun
> プロセスの環境変数が見えるようなので、これで落ちるプロセスの環境変数が適切か見てれるかな >

https://github.com/start-jsk/rtmros_hironx/issues/539#issue-598511308 のipythonコマンドを実行後、そのpythonのものと思われるプロセスの環境変数を確認したところ、以下のようになり、`ORBgiopMaxMsgSize`は見当たりません。
```
$ ps -a                                                                                                       
  PID TTY          TIME CMD                                                                                                          
   37 pts/0    00:00:00 su                                                                                                           
   38 pts/0    00:00:00 bash                                                                                                         
   68 pts/0    00:00:00 tmux: client                                                                                                 
  191 pts/1    00:00:04 rtmlaunch                                                                                                    
  195 pts/1    00:00:00 omniNames                                                                                                    
  695 pts/2    00:00:04 python                                                                                                       
  746 pts/3    00:00:00 ps                    
$ sed -e 's/\x0/\n/g' /proc/695/environ |grep ORBgiopMaxMsgSize                                                                                       
$ sed -e 's/\x0/\n/g' /proc/695/environ
TMUX=/tmp/tmux-1000/default,70,0                                                                                                     
MAIL=/var/mail/pazeshun                                                                                                              
USER=pazeshun                                                                                                                        
HOSTNAME=46405e9dcc2a                                                                                                                
EUSDIR=/opt/ros/melodic/share/euslisp/jskeus/eus/                                                                                    
ROS_PACKAGE_PATH=/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/hiro_card_demo:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robot$
cs/rtmros_hironx/hironx_calibration:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models:/home/pazeshun/ros/j$
k_demo_ws/src/rtm-ros-robotics/rtmros_hrp2/jsk_models:/home/pazeshun/ros/jsk_demo_ws/src/multisense/multisense_description:/home/paz$
shun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_nextage/nextage_description:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtm$
os_nextage/nextage_gazebo:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_nextage/nextage_ik_plugin:/home/pazeshun/ros/js$
_demo_ws/src/rtm-ros-robotics/openrtm_aist:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/openhrp3:/home/pazeshun/ros/jsk_demo_$
s/src/rtm-ros-robotics/hrpsys:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_common/hrpsys_tools:/home/pazeshun/ros/jsk_$
emo_ws/src/rtm-ros-robotics/rtctree:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_common/rtmbuild:/home/pazeshun/ros/js$
_demo_ws/src/rtm-ros-robotics/rtmros_hironx/rtmros_hironx:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_nextage/rtmros_$
extage:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtshell:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtsprofile:/h$
me/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_common/openrtm_tools:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtm$
os_common/hrpsys_ros_bridge:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_hironx/hironx_ros_bridge:/home/pazeshun/ros/j$
k_demo_ws/src/rtm-ros-robotics/rtmros_hironx/hironx_moveit_config:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_tutoria$
s/hrpsys_ros_bridge_tutorials:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_nextage/nextage_ros_bridge:/home/pazeshun/r$
s/jsk_demo_ws/src/rtm-ros-robotics/rtmros_nextage/nextage_moveit_config:/home/pazeshun/ros/jsk_demo_ws/src/rtm-ros-robotics/rtmros_n$
xtage/nextage_calibration:/home/pazeshun/ros/jsk_demo_ws/src/turtlebot/turtlebot_description:/home/pazeshun/ros/jsk_demo_ws/devel/:/$
pt/ros/melodic/share                                                                                                                 
LD_LIBRARY_PATH=/opt/ros/melodic/share/euslisp/jskeus/eus//Linux64/lib:/home/pazeshun/ros/jsk_demo_ws/devel/lib:/opt/ros/melodic/lib$
/opt/ros/melodic/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/melodic/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/melodic/share$
euslisp/jskeus/eus//Linux64/lib                                                                                                      
ROS_ETC_DIR=/opt/ros/melodic/etc/ros                                                                                                 
SHLVL=3                                                                                                                              
HOME=/home/pazeshun
ROS_PYTHON_VERSION=2
ROS_DISTRO=melodic
LOGNAME=pazeshun
_=/usr/bin/ipython
ROS_VERSION=1
PKG_CONFIG_PATH=/home/pazeshun/ros/jsk_demo_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig
TERM=screen
RTCTREE_NAMESERVERS=localhost:15005
[... cut ...]
01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v
=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=
01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;
36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;3
6:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
TMUX_PANE=%0
```

YoheiKakiuchi, 2020-04-22:
問題の理解が足りてなかった気がしますが、
size指定を片側だけにすると、16.04ですが、やはり落ちるので、model-loaderだけの設定で上手く行っていた方がomniのpythonのバグかもしれませんね。
```
rosrun openhrp3 start_omninames.sh 9999 &
sleep 1
ORBgiopMaxMsgSize=2147483648 openhrp-model-loader -ORBInitRef NameService=corbaloc:iiop:localhost:9999/NameService &
sleep 1
hrpsys-viewer $1 -ORBInitRef NameService=corbaloc:iiop:localhost:9999/NameService
```

PR #540 (2020-04-23): force set ORBgiopMaxMsgSize=2147483648  for hironx_client.py
hot fix for https://github.com/start-jsk/rtmros_hironx/issues/539

On 16.04 (omniorb4-dev 4.1) if we start openhrp-model-loader with <env name="ORBgiopMaxMsgSize" value="2147483648" /> all connection(?) conect respenct giopMaxMsgSize
 But on 18.04 (omniorb4-dev 4.2) wnneed to set ORBgiopMaxMsgSize=2147483648 for each clients

Closes #539
closed 2020-04-23 by k-okada